betaflight/betaflight-configurator #1134: Feature Request - motor resource mapping via Configurator

kenkamm, 2018-07-25:
I love being able to remap motors via CLI commands - I never worry about how I wire up my 4-in-1 ESCs, I just remap the motors as needed afterwards. However it sure would be nice to be able to do this in the GUI. 

4-in-1 ESCs are becoming more and more popular, and this would also make it easier to remap motors when a AIO FC isn’t installed in the normal orientation.  

Perhaps a drop-down box could be added on the motors tab, under each slider, so that the actual motor spinning could be chosen. When saved, the motors would remap appropriately. 

Thanks,
Ken

Dave9669, 2018-08-01:
I think that is a great idea

fujin, 2018-08-01:
RF/FL1's wizard here is really nice. It spins 'em one by one and asks you which motor position it is. Then it runs again to check rotation and uses dshotprog to normal/reverse/save. I think a similar approach would work well.

kenkamm, 2018-08-01:
Agreed, the RF1 approach is quite nice. It also has a wizard to determine FC orientation. But I was trying to think of a solution that would be easy to implement. I'll bet the RF1 implementation took a lot of time.

wolkesson, 2018-08-15:
Also being able to reverse the motor (via dshotprog) would be fantastic. That's the only think I use BLHeli configurator for today. The RF1 wizard is nice, but probably took a lot of time to implement. But dropdowns for output selection and checkboxes for reverse (if using dshot) would be a nice start. Should it be on the motors tab or the configuration tab (where reverse all is today)

mikeller, 2018-08-16:
@kenkamm: Sounds like a good plan - now there just needs to be somebody with the time to implement it. :wink:

dthubereng, 2019-02-23:
@mikeller: this request is still open and I may have some time to work on this if it's still desired. Need to wrap my head around it first.

mikeller, 2019-02-23:
It's definitely one we'll have to tackle at some point.
Probably best if you hold off some more weeks, there are changes under way how resources / timers / DMA is done in the firmware, and it will be best to wait with specifying the MSP API for these things until the changes are done.

julie777, 2019-04-24:
This would probably eliminate 90% of the times that I must use blhelisuite. I am all for it.

kenkamm, 2019-04-24:
Maybe now that 4.0 has been released there would be time for the devs to work on this. Or maybe not - supporting the latest release is a lot of work. Either way, I look forward to this being implemented, if it’s still an option.

mikeller, 2019-04-28:
Yeah, now's good, if you are still interested @dthubereng. :-)

dthubereng, 2019-04-28:
Ok, I will look into it when I get back to the states in a couple weeks. :)
    	
    	

    	Dave HuberPrincipal Engineer - Wheel LoadersKomatsu Mining Corp. GroupP.O. Box 2307Longview TX 75606-2307United StatesOffice: +1 903 237 7148Mobile: +1 903 619 0155

kenkamm, 2019-08-09:
@dthubereng any luck yet? This would be really helpful. ;)

dthubereng, 2019-08-09:
@kenkamm I am so sorry! Life got the better of me these last few months and I've started doing a lot of extra programming at work (so home time has been spent with the family). I haven't forgotten about this, though. Still in my inbox reminding me on an almost daily basis. Let me see if I can get back into it this weekend and try to make some headway on it. I may have some questions about implementation. Again, sorry for dropping the ball on this for so long.

kenkamm, 2019-08-09:
@dthubereng certainly no need to apologize. Not like I can code it myself! I don’t have those kinds of skills. So, thanks in advance if you do find some time to implement it. If you don’t have time, no worries.

dthubereng, 2019-08-21:
Well, it doesn't look like I'm going to be able to get to this for a bit. It'll stay on my radar and if I get a free weekend, I'll def look at it if someone else hasn't done something. I just can't seem to etch out a solid segment of time to give it the attention it needs. I'll let you know for sure if I do.

haslinghuis, 2022-01-19:
Fixed in:
https://github.com/betaflight/betaflight-configurator/pull/2093
https://github.com/betaflight/betaflight-configurator/pull/2296

closed 2022-01-19 by haslinghuis