betaflight/betaflight-configurator #1136: GPS datas aren(t export in CSV file

LeDoudou, 2018-07-26:
Does it possible to add GPS raws datas into .CSV files ? (Betaflight 3.4)

The datas below seems to be recorded : 
==============================

#ifdef USE_GPS
// GPS position/vel frame
static const blackboxConditionalFieldDefinition_t blackboxGpsGFields[] = {
    {"time",              -1, UNSIGNED, PREDICT(LAST_MAIN_FRAME_TIME), ENCODING(UNSIGNED_VB), CONDITION(NOT_LOGGING_EVERY_FRAME)},
    {"GPS_numSat",        -1, UNSIGNED, PREDICT(0),          ENCODING(UNSIGNED_VB), CONDITION(ALWAYS)},
    {"GPS_coord",          0, SIGNED,   PREDICT(HOME_COORD), ENCODING(SIGNED_VB),   CONDITION(ALWAYS)},
    {"GPS_coord",          1, SIGNED,   PREDICT(HOME_COORD), ENCODING(SIGNED_VB),   CONDITION(ALWAYS)},
    {"GPS_altitude",      -1, UNSIGNED, PREDICT(0),          ENCODING(UNSIGNED_VB), CONDITION(ALWAYS)},
    {"GPS_speed",         -1, UNSIGNED, PREDICT(0),          ENCODING(UNSIGNED_VB), CONDITION(ALWAYS)},
    {"GPS_ground_course", -1, UNSIGNED, PREDICT(0),          ENCODING(UNSIGNED_VB), CONDITION(ALWAYS)}
};

// GPS home frame
static const blackboxSimpleFieldDefinition_t blackboxGpsHFields[] = {
    {"GPS_home",           0, SIGNED,   PREDICT(0),          ENCODING(SIGNED_VB)},
    {"GPS_home",           1, SIGNED,   PREDICT(0),          ENCODING(SIGNED_VB)}
};
#endif

mikeller, 2018-07-27:
@LeDoudou: This is already possible:

[image]

LeDoudou, 2018-07-28:
Hello @mikeller thanks.
Does the message means : 
 1. I connect my battery
 2. I start betaflight and select msp_raw_gps
 3. I don't go out of this tab, I just disconnect the usb cable and start to flight ?

mikeller, 2018-07-28:
@LeDoudou: No, you need to stay connected to the configurator, otherwise the configurator won't be able to record logs, obviously.

LeDoudou, 2018-07-28:
Sorry, but I would like to record GPS datas of the flight

mikeller, 2018-07-28:
@LeDoudou: The Betaflight configurator won't be able to do this, since it needs a connection to the flight controller to be able to receive data.

closed 2018-07-28 by mikeller

LeDoudou, 2018-07-28:
I think there's a misunderstanding.

Does it possible to : 
 1. Record gps datas in the blackbox during the flight.
 2. Be able to watch (or export) this datas in Betaflight Blackbox Explorer ?