betaflight/betaflight-configurator #1139: Feature Request - A Range of Presets

bigd942, 2018-07-29:
With the changes in 3.4 and 3.5 and to the configurator 10.4 things seem to be getting slightly more complex. I like to 'just fly' but still like to tweak a little to get the characteristics I prefer. The tuning tips for 3.4 and 3.5 are really good but can these be rolled into selectable presets, possibly even so I can switch between them via OSD/AUX to see how much a difference they make. In the real world a 'clean build' can quickly become a 'not so clean build' so it would be good to change the settings quickly for filtering and dynamic notch. Thanks team for all your efforts!

BradB111, 2019-06-26:
Almost a year later and I think this would still be a brilliant idea. under the pid tuning tab there could be a 'sub tab' for community presets from https://github.com/betaflight/betaflight/wiki/Community-Presets
where a brief description of the preset can be included and its intended use case. 

At the very least it could become a good base tune for similar quads. e.g. a 2" micro tune preset may not be 100% suitable for a specific 2" build but would serve a better starting point than the current defaults which seem to be more suited to 5".

With the amount of tuning difficulty some people seem to be experiencing with 4.0.x this may resolve some of the issues by providing different tunes the user could try without needing to spend hours learning exactly how to tune a quad.

haslinghuis, 2024-05-06:
We have a nice preset system now

closed 2024-05-06 by haslinghuis