betaflight/betaflight-configurator #1140: betaflight configurator cannot to dump all

venom-code, 2018-07-30:
Betaflight 10.3.1 when trying to do a dump all in cli command the programme stalls and gives a half dump and then doesnt function until usb is disconnected from fc and configurator is restarted, tried on multiple quads/fc's same result of stalling.
using windows 10

mikeller, 2018-07-31:
@venom-code: This typically happens when the flight controller configuration is corrupted after a firmware upgrade was done without resetting the configuration. Reset your configuration to defaults, and then restore it from a backup.

kenkamm, 2018-08-16:
I’ve had this happen recently too. I always do full chip erase and full reset before pasting my diff in. I reflashed and the issue went away, but this was a new one for me.

d3nyys, 2018-08-27:
I investigated further why this is happening and I think it's a bug in Betaflight Configurator 10.4.0

If I reset the controller and imediatly run the dump all CLI command it works, but as soon as I press Save and Reboot(even if I don't make any changes) in the config tab something breaks and the dump all command freezes inbetween displaying the data.

I tested the same thing in 10.3.1 configurator and it doesn't happen. Also if I make some changes in the CLI and use the save command it works ok and the dump command works afterwards.

venom-code, 2018-08-27:
i again reset everything and manually did all my preferred settings but the issue persists, just did a new build with a KakuteF4 and its also having the same dump and diff issues, i tried to backup the factory settings with a diff but the configurator again freezes, i cannot do a dump all or a diff to backup anything, when i have time i will uninstall Betaflight configurator and reinstall it and will post results.

dingrgh, 2018-08-28:
This happened to me as well a week ago. After making some change in the Configuration tab and saving, going into CLI and doing a dump or diff resulted in this behaviour. I think it was something in the 'Beepers' section of the Configuration tab but not I'm not sure.

The problem went away after I did the following and has not happened since. But I cannot confirm for sure what I actually did that fixed it. Here's what I did:

Connect USB to power up the FC.
Start up Configurator. 
Go to the CLI tab immediately.
Type 'defaults'.  <-- This will reset everything to defaults and you will have to configure from scratch.
Configure parameters as per normal with the exception of the following... When in the Configuration tab, make only ONE change at a time, saving after each change.

Since it all works for me now, I have not done any trouble shooting to narrow down the problem. Perhaps someone elses can narrow the problem down to something more specific.

mikeller, 2018-08-28:
The only potential culprit I can see is #995. But since this was added before 10.3.0, the fact that some users report this not happening with 10.3.0 makes it unlikely...

@ademuk: Do you have any idea what this could be?

ademuk, 2018-08-30:
@mikeller nothing springs to mind configurator wise.

Are there any JS errors in the console when the error occurs?

Have there been any changes to the way the firmware outputs the dump/diff?

ademuk, 2018-08-30:
I'm unable to reproduce this.

I tried with MacOS, 3.4.0 configurator, on both 3.2.2 + 3.5 firmwares. 

Can we get the following:
* Operating system.
* Firmware version
* Configurator version

d3nyys, 2018-08-31:
Win 10 Pro 1803
Configurator 10.4.0
Betaflight 3.5

The problem is in Save and Reboot on the Configuratin tab. On the Ports tab everything is ok, and every other normal Save on every other tab works ok too. Save & Reboot from CLI works without problem too.

wes1007, 2018-09-01:
Im also seeing a similar issue

Win 10 Pro 1803
Configurator 10.4.0
Betaflight 3.5
Flight Controller: Betaflight F4

`Diff` Locks up the CLI instantly. `Dump` seems to Also lock up part way though. GUI is still responsive and i can scroll through the CLI but I cant interact at all with the Flight controller. only way to recover from the locked up state is to power cycle the Flight Controller
Steps to reproduce:
Flash BF3.5 with full chip erase
flash completes successfully
connect to Flight controller and run Diff in cli (works as expected)
Change ESC protocol in configuration tab then save and reboot (BF configurator seems to hang at this stage. I have to disconnect from FC and reconnect)
Go to CLI type Diff and then the CLI hangs. Disconnect only works sometimes but even then the FC seems to be stuck in CLI and has to be power cycled.

Running through the same process with BF 3.4.1 doesnt have the same issue. This man be a Firmware issue and not a Confiurator issue

mikeller, 2018-09-01:
Ok, I had an instance of this.

It looks a lot like it's a compound problem: It's a firmware config corruption issue, causing a `diff` (or `dump`) to return parameters up to one where it locks up trying to dump a very long string as the result of a value lookup table index being out of bounds, trying to read 0 terminated string outside of a string table. This is a known problem with corrupted configs and has always been around.

The second part of the problem is that this very long string results in an exception in `cli.js` when trying to parse the output, causing the configurator to go into a loop and hard lock until it is restarted. This seems to be new and potentially caused by the new code to do tab completion.

Unfortunately the problem was not reproducible - I reset the configuration, and applied the config I had before the firmware update, and the problem was gone. I also did not get any stack traces before I attempted to reproduce it.

ademuk, 2018-09-03:
I've been able to reproduce this on Windows 10 with 3.5 and 10.4.0 configurator.

See screenshot below:
[image]

I've also tried with https://github.com/betaflight/betaflight-configurator/pull/995 reverted. The error still occurs.

Happy to help with this issue. I'll try and dig deeper this week.

mikeller, 2018-09-03:
@ademuk: Thanks for looking into this. Looks like you found a different error to what I got, mine was a 'regex parsing failed' or similar, and it was repeating and counting up at a rate of 100s / second.

ademuk, 2018-09-09:
Is this fixed by https://github.com/betaflight/betaflight/pull/6696 ?

edifus, 2018-09-09:
Still having this issue with firmware 3.5.1 and configurator 10.4.0

Win 10 Enterprise 1803
Configurator: 10.4.0
Betaflight: 3.5.1
Flight Controller: BETAFLIGHTF4

Unable to run 'dump' or 'dump all' at the CLI, output doesn't complete and stops part way through.

I don't have the issue with a CLRACINGF4 board. Wondering if I should open an issue on the firmware tracker or if this is still a configurator issue.

mikeller, 2018-09-10:
@edifus: Can you please reset your configuration to defaults and then try to replicate?

edifus, 2018-09-11:
@mikeller Seems to have done the trick this time around, no longer hanging with dump/dump all. I always do a full chip erase when I flash so not sure what's different this time. Thanks!

closed 2018-10-18 by stale[bot]