betaflight/betaflight-configurator #1148: Dataflash download not working with X220

ir-fuel, 2018-08-06:
I am running Betaflight 10.3.1, firmware 3.4.1 (Target:SRF3). The application works perfectly on my mac, however downloading the dataflash to a file from my Eachine X220 freezes at 0%. Nothing happens and the application hangs.

Any idea on how I can figure out what the problem is? I am a (web) developer myself so no problem getting technical.

Thanks!

jflyper, 2018-08-12:
OSX doesn't like how BF emulates a FAT32 file system atm.

3.4 MSC for onboard flash does not work on OSX. #6307

AndersHoglund, 2018-08-12:
Could this perhaps be a duplicate of https://github.com/betaflight/betaflight/issues/6348

Philipameen, 2018-08-13:
I use Betaflight 10.3.1 on osx High Sierra and have the same problem, process is stuck at 0%. Im running firmware 3.4.1. I can get it to save properly but only if i downgrade firmware (I tried with 3.0.1).
My FC is a D-Link F3 V3.

closed 2018-10-09 by stale[bot]