betaflight/betaflight-configurator #1151: [bug] configurator 10.4 hangs on "Ports" tab

libwtf, 2018-08-11:
configurator 10.4 from http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Configurator/ (build261) hangs when going to "Ports" tab.

> Firmware upgrade required. Serial port configurations of firmware < 1.8.0 is not supported.

i can set ports settings thru CLI.

[image]

thanks!

carloscampos775, 2018-08-11:
Same Message Here

Firmware upgrade required. Serial port configurations of firmware < 1.8.0 is not support

Using a Betaflight F4

mikeller, 2018-08-12:
Configurator 10.4.0 is not released yet. Please use an official build.

closed 2018-08-12 by mikeller

FrozenDroid, 2018-08-12:
You could've mentioned that it was indeed a bug. I can see it was fixed in #1152.
We were pointed to that build server by JB.

frozenskys, 2018-08-12:
@FrozenDroid It's not a really bug when that version of software isn't released - if you are using un-official developer builds (not beta or RC builds) of software it may or may not work. JB should have been clear in his video that these are not even beta quality builds (they are just built off the latest version of the code) - and that it may have incomplete\buggy changes so YMMV.

mikeller, 2018-08-13:
@libwtf: I never said it was not a bug - I just stated that you were reporting a bug for a version that does not exist, and should consider using a released version unless you are specifically testing configurator and are willing to use known buggy software and are willing to risk your configuration being messed up. (In this case, can you please start attaching console output to your bug reports, this will make them a lot more useful.)