betaflight/betaflight-configurator #1163: Not Connecting in Betaflight 3.4.0

Hotdogs4, 2018-08-14:
I received the notification that Betaflight needed to be updated, so I went to the GitHub page. But, when I pressed the download button for Windows 10, it would fail and say network error. Then, I tried the chrome download, and it worked, but the connect button in the top right wouldn't even press, and was grey. Does anyone know why this is?

ED-weed, 2018-08-14:
"No configuration data within 10 seconds, communication failed" is the msg I get now when trying to connect.  What's the deal???

mrshutterbug, 2018-08-27:
I updated to 10.4.0 and I get the same issue. Is there a work around? I am running the stand-alone version for MacOS X. The "connect" button remains grey and I cannot click on it. If I select "auto connect" before plugging in the USB cable, then it will change to "Connecting" but it still doesn't work.  The log says:

2018-08-27 @ 08:13:59 -- Running - OS: MacOS, Chrome: 67.0.3396.99, Configurator: 10.4.0
2018-08-27 @ 08:13:59 -- Using cached release information for configurator releases.
2018-08-27 @ 08:16:33 -- Serial port successfully opened with ID: 8
2018-08-27 @ 08:16:43 -- No configuration received within 10 seconds, communication failed

If I revert to version 10.3.0 of the configurator, then it works perfectly. Please help.

mikeller, 2018-08-28:
@mrshutterbug: At least knowing what version of MacOS X that you are getting the problem with would be helpful.

mrshutterbug, 2018-08-28:
Sorry. It's Mac OS version 10.11.6