betaflight/betaflight-configurator #1170: Betaflight Chromebook

Helsbee, 2018-08-15:
Betaflight no longer works on my chromebook.
The button to connect is grayed out, doesnt even try to load on firmware update.
checked cleanflight which seems to still work fine....Help me anyone?

fujin, 2018-08-15:
It's already been fixed (https://github.com/betaflight/betaflight-configurator/pull/1168), 10.4.1 version should appear in the Chrome store at some point, or you can use this one manually: https://ci.betaflight.tech/job/BetaFlight_Configurator/60/artifact/release/betaflight-configurator_10.5.0_chromeos.zip

closed 2018-08-15 by fujin

Helsbee, 2018-08-15:
ive just installed betaflight dyd from the chromestore is this the same or if not....what is the difference?

mikeller, 2018-08-15:
@Helsbee: Just keep in mind that, if you force install the app from a file, updates will be broken, and you will have to scrub the local version from your install, and install the chrome web store version in order to get updates to the application to work again.

fujin, 2018-08-15:
@Helsbee the version in the chrome store has the bug. The one that isn't in the chrome store yet has been fixed. :) as @mikeller mentioned, patience is probably the best choice here. When 10.4.1 makes it to the chrome store you can get it from there.