betaflight/betaflight-configurator #1171: Betaflight Configurator 10.4.0 becomes unresponsive on Chromebook

SlimPicket, 2018-08-16:
Chromebook Chrome OSVersion 67.0.3396.99 (Official Build) (64-bit).
Betaflight Configurator 10.4.0, August 13 
Successfully installed from webstore
After starting the App the Connect button stays grey.
Firmware Flasher menu hangs on status "Loading".
Unable to swich back to menu items Welcome or Documentation & Support
Unresponsive for other actions except closing the App.

vayulove, 2018-08-16:
I think I have the same issue on debian sid/KDE with Chrome version 68.0.3440.75 also 68.0.3440.106.  Betaflight 10.4 from the Chrome store installs, but when I run it, and connect a board the device shows the proper USB port but the connect button is unresponsive.  Then if I go to the firmware flash section, the dropdowns are stuck on "loading...", and doesn't give me a choice of firmware.  Also once I go to the firmware flash section the program seems hung and won't allow me to switch to the Documentation nor Welcome tabs.   

I have 10.4 running fine from a .deb file.

salty511, 2018-08-16:
Same issue here on windows 10, Chrome version 68.0.3440.106, Betaflight Config version 10.4.0, installed from Chrome store.
Connect button is grey'd out from launch however connecting my board via usb causes it to show the correct port (COM4) but no change to connect button. Switching to firmware flasher tab, drop down menus both hang on loading, clicking load firmware [local] does nothing and I'm unable to get back to the other menus

Log
2018-08-16 @ 15:49:21 -- Running - OS: Windows, Chrome: 68.0.3440.106, Configurator: 10.4.0
2018-08-16 @ 15:49:21 -- Using cached release information for configurator releases.
2018-08-16 @ 15:49:27 -- Using cached builds information for jobs.

BlackhawkFPV, 2018-08-16:
Same issue as salty511.  Haven't used Bflight in several days, got on today and button is grey, can not connect, but shows correct com port.  Downloaded the actual BFlight standalone program and works fine.  Found the standalone program here:  

 https://github.com/betaflight/betaflight-configurator/releases

And was linked from this article:

https://github.com/betaflight/betaflight/issues/6562

Using Chrome, windows 10 64b, bflight 10.4

Was able to open bl heli suite no problem with chrome app opened, but still grey connect button in chrome bflight app.

FReeSKYmagnumFLY, 2018-08-19:
Same issue as OP SlimPicket. Running ChromeOS on a Samsung Chromebook. 

Every attempt made to subvert this issue has lead swiftly to disappointment.
I was able to connect and use every aspect of BF just a day ago (Aug 17th 2018), yet today (Aug 18th) the connect button is grey and unresponsive. As well as other issues stated by those before me.

There doesn't seem to be a standalone version available for the ChromeOS either, only Windows, Linux and Mac. I am in desperate need of a configurator, and completely out of ideas.

mikeller, 2018-08-19:
#1169 has fixed the issue - it just needs to be deployed to the Chrome web store for the update process to start. I hope @borisbstyle will get around to pushing this into the store soon.

Sven062814, 2018-08-19:
<<<Another frustrated chrome user here, who has tried every possible combination of installing/ uninstalling. We are all just dead in the water until this fix becomes available to us.

kzvt, 2018-08-19:
Same here.
Chrome 68.0.3440.106 
Betaflight 10.4.0
OS: Windows 10 pro

Btw it is really annoying that you can't downgrade betaflight to older (working) version.

mikeller, 2018-08-19:
@Sven062814: Are you using Chrome, or a chromebook?

@kzvt: If you are on Windows, you should not be using the Chrome web app version in the first place, but the stand alone version for Windows. In the future, please read the release notes. They tell you loud and clear at the very top to use a different version to what you are trying to use:

https://github.com/betaflight/betaflight-configurator/releases/tag/10.4.0

Sven062814, 2018-08-19:
HP Chromebook. I'm pretty much computer illiterate when it comes to anything beyond the basics. I appreciate the link, but its my understanding, and the fact that I tried anyway, that none of those versions will open on HP Chromebook. 
Thank you for the response

mikeller, 2018-08-19:
@Sven062814: The fix for Chromebooks is scheduled to be released as soon as @borisbstyle gets around to pusing it to the Chrome web store - we are all doing this in our spare time...

Sven062814, 2018-08-26:
Betaflight is now working again. Thank you again guys, what you do is huge for the hobby!!!

FReeSKYmagnumFLY, 2018-08-30:
Thank you so much! Sorry about the lag response.

SlimPicket, 2018-09-01:
Issue is fixed after installing version 10.4.1 August 24 2018.
Big thanks to the devs!

closed 2018-09-01 by SlimPicket