betaflight/betaflight-configurator #1179: Feed Forward not saving changed value on yaw axis.

tekstyle1, 2018-08-25:
Hey guys I cant save the Feed Forward value in config 10.4.0 on the yaw axis. It shows as 0 from default . It works great on roll and pitch but not yaw. Tried to save in cli with the same result. Tried on multiple FCs with same result. (BFF7, Zeus, Matek f722) Tried reinstalling Config a couple times with new copy and same result. Same on multiple pcs .. am I missing something ?

tekstyle1, 2018-08-25:
set the recommended set iterm_relax = RP
set rc_smoothing_type = FILTER
set rc_interp = AUTO
set rc_interp_ch = RPYT
and it seems to save the yaw FF now.? weird but its saving so im good i guess . worked on 2 quads lol.

closed 2018-08-25 by tekstyle1

mikeller, 2018-08-25:
@tekstyle1: if you'd posted a config `diff` like you were asked to do when opening an issue, then maybe somebody would have told you why this did not work for you.

Feedforward can only work on smoothed inputs, and from the look of it your yaw input was not smoothed.