betaflight/betaflight-configurator #1183: Feature request - one-click "save" dump, diff, and screenshots automatically of all config pages

johnlenfr, 2018-08-28:
Like discuss in Inav would be great feature to have the ability to auto-save all of the configuration in an Zip file that countain:
- a "dump" in a text file
- a "diff" in a text file
- screenshots of all config pages (each page to a different jpg file) 

https://github.com/iNavFlight/inav/issues/3590
https://github.com/iNavFlight/inav-configurator/issues/544

Like @VoicOfReason suggested : "For simplicity, the export could be a zip file with craft name, date, FC
name, and iNav firmware version as file name."

d3nyys, 2018-08-28:
I'd like that too. I do the same procedure manually.

timman2er, 2018-10-09:
We could use the backup and restore button that already exists for this, as the backup button would be pretty useless if this were added.

The backup button would do the function described in OP, in order to restore you'll update either your dump or diff file and everything gets restored.

Another thing to consider is adding a warning before flashing about making a backup, but that is a thing for a seperate issue.

MicroMontis, 2019-01-03:
Cool feature. I do copy-paste the DIFF text and save it to file everytime.

closed 2024-05-06 by haslinghuis