betaflight/betaflight-configurator #1191: OSD configurator for 16:9

McGyverHenke, 2018-09-03:
Im wondering if it is possible to add a feature that makes it possible to see OSD in a 16:9 aspect ratio. When I edit and organize the elements i have to get my goggles and that is quite troublesome. If there just were a switch where i could toggle between 4:3 and 16:9 aspect ratios. :)

frozenskys, 2018-09-04:
Not sure how this would work as we could obviously scale the OSD picture in the configurator to show in 16:9 mode but I'm not sure how close this would be to the actual picture in the goggles. The OSD is capable of 30x13 (NTSC) or 30x16 (PAL) characters and has no adjustment for 4:3 or 16:9 aspect ratios - I think this would depend on how the goggles / camera work together? a 4:3 camera signal to a 16:9 dispaly will crop differently to a 16:9 camera signal to a 16:9 display, etc.

Docteh, 2019-04-03:
I've noticed that different goggles and televisions crop the image in different places. A configurable stencil that masks out a certain part of the display as a reminder to the user could be a useful option, but then where would you save that information? The OSD has Horizontal and Vertical offsets to help align things. Actually a display calibration mode could be entertaining

TyLauriente, 2022-12-09:
Now that digital systems like HDZero, Walksnail and DJI have OSD capabilities. It seems like this feature is much needed. Currently (using Walksnail) I can't extend the OSD to the edge of the video feed. It appears as a square in the middle of my image. Inav has figured this out already but I prefer befaflight for flight performance. Please implement this feature!

BoldBigflank, 2023-06-23:
I believe this was added in https://github.com/betaflight/betaflight-configurator/pull/3071

atomgomba, 2023-06-23:
> I believe this was added in #3071

Since this is an issue from 2018 I'm pretty sure OP means SD 16:9 which is supported by PAL (AFAIK NTSC can only do 4:3)

closed 2023-06-23 by haslinghuis