betaflight/betaflight-configurator #1195: built in updater

frizzle101101, 2018-09-09:
please can someone help get this feature started for the configurator

mikeller, 2018-09-10:
@frizzle101101: The title and text of your feature request are not clear at all. Can you please clarify or close?

frizzle101101, 2018-09-10:
Sorry i mean like when a release is released it goes up on the page as well as into an auto update database or something. Then a built-in updater in the configuration will alert the user of and an update and do everything for them so they do not have to download and install the new version every time

timman2er, 2018-09-20:
Now I'll have to go download the new version from the release page and will end up having multiple versions of the bf-configurator installed, so this would be nice.

timman2er, 2018-10-09:
What I think he means is an update for the configurator. In order to update you'll have to remove the old configurator (otherwise you'll get multiple versions of the configurator) right now and download a new one. What'd be better is having it auto update its self.

McGiverGim, 2018-10-09:
The actual installer (at least Windows/Linux) removes the older installation before installing the new version.

mikeller, 2018-10-10:
@timman2er: Auto-update, and in particular the ability to update by only pulling the updated js / html / css files for the configurator and not the 10s of MB of NW.js that they are wrapped in would be great.

timman2er, 2018-10-10:
What will happen on the of chance a webkit needs an update though?

timman2er, 2018-10-10:
@mikeller for me (or probably anyone else interested in adding this), it might be handy to list how updates are currently done (what server etc.).

mikeller, 2018-10-11:
@timman2er: Currently we only do full updates, packaged as installable packages / executables, and they are distributed through GitHub - so for automatic updates we'd probably have to come up with a new way to distribute. Maybe we could use the http://betaflight.com/ server that is provided and managed by @wind0r.

timman2er, 2018-10-11:
We're talking about auto update, but to be sure it`s user initiated by a press, after which everything is done automatically for them right?

timman2er, 2018-10-11:
Also, there seems to be documentation available for this already: http://docs.nwjs.io/en/latest/For%20Users/Advanced/Autoupdates/ .

There seem to be two we can choose from:

- https://github.com/oaleynik/nwjs-autoupdater (doesn't require elevated priveleges)
- https://github.com/dsheiko/nw-autoupdater (Much more active and better documentation)

I think we'd best go with nw-autoupdater.

timman2er, 2018-10-13:
Well I think that modifying it based on only changed files would be best. It'd require to check a "last modified" timestamp like you have in windows folders and only changing those files.

mikeller, 2018-10-14:
Even if all the source files for the configurator were pulled in for every version change this would still reduce the size of the download by ~ 90% (~ 7MB source files vs. ~ 70MB for the packed NW.js installer).

timman2er, 2018-10-14:
Good point, but I think the following needs to be considered:

- Will any nw.js files be updated?
- We also have resource files etc. and they get updated too (although infrequent)

mikeller, 2018-10-14:
- The nw.js files are only updated when nw.js is updated to a new version (or so I assume, the toolkit and source files are rolled into an application by the nw-builder);
- Good point about the resources, they make up another 15 MB. But even that is small compared to the size that the nw.js toolkit adds.

I just did a quick test, and the combined size of `src/` and `resources/` in a zip file is ~ 8 MB.

timman2er, 2018-10-14:
Well from the first comment today it sounds like you just want to pull src and resources, making that 8 mb. If that's so, how do we deal with nw.js updates?

mikeller, 2018-10-14:
The way we do until now - we ask the user to download / install the updated version - there is not really a way around that anyway.

timman2er, 2018-10-14:
I think that updating files based on whether it's modified or not would be a simple enough step from what's out now. If we were to move to more frequent updates, I think we should do so. It'd make a nice Github repo as well. This should make updating much less painful.

mikeller, 2018-10-15:
How would you detect if a file has been modified, short of downloading and comparing the files?

mikeller, 2018-10-15:
@McGiverGim: That would work.

But the effort for generating this file as part of the build process, and then hosting it and all of the files for every release, and doing the comparisons and on-demand downloads will be considerable.

wind0r, 2018-10-15:
For updating electron apps there is https://github.com/GitbookIO/nuts and https://github.com/zeit/hazel .. which just use github releases as a backend.

maybe there also is something similar that just can be used for nw.js. then we do not need to do some homegrown solution. just updating the js files should be small enough.

McGiverGim, 2018-10-15:
True, the problem here is to store a tree with all the versions, for each operating system, in a server, maintaining this.

To me is not a problem to execute the complete installer for each version ;)

wind0r, 2018-10-15:
Ohh. Missed a post above where those soltuion for nwjs allready got named.

I dont get why a tree needs to get stored. Github releases is just a list. Updates on file based isnt needed for that small about of kB. The Updater just will download the newest html/css/js bundle zip

@mikeller just a small correction betaflight.tech is manged by me ;) blckmn has the control for betaflight.com. he also owns betaflight.tech but added it to my nameserver.
When needed i can host nw-autoupdater under conf-update.betaflight.tech (or similar)

timman2er, 2018-10-15:
@wind0r but then how will nw.js be updated?

mikeller, 2018-10-16:
@McGiverGim: The problem is how often we can release a new version without user complaints. Back when we only had the Chrome Web Store version, we could release an updated version every month or so, and nobody ever complained. Now we have already had complaints about releasing too often...

@wind0r: Thanks for the correction. I meant more the server that both domains are pointing to at the moment, as a potential place to host files - what domain we use for the URL to refer to these files will be another question. :-)

closed 2018-11-22 by stale[bot]