betaflight/betaflight-configurator #1200: Control Sticks function isn't working for armed mode

weixingsun, 2018-09-19:
I want to use configurator to simulator the real remote control.
already in latest version configurator 10.4.0 and betaflight 3.5.1 , on Configuration tab, set the Receiver option to "MSP RX input (control via MSP port)", save & reboot, 
I can use Motors tab to control the speed of motors, every of motors is working properly
and switch to receiver tab, the motors keep spinning
but I am expecting the motors change speed when I play with control sticks (clicked on enable controls button)

Is there any option to be enabled in Receiver tab ?
Now I can only see the Preview animation changes, but I want to see the real armed motors spin
Please help, thanks

weixingsun, 2018-09-20:
Any suggestion? Thanks in advance

weixingsun, 2018-09-20:
I enabled "Motor Test Mode", otherwise the motors won't spin
Why don't you think it may be a defect?

weixingsun, 2018-09-20:
thanks in anyway

mikeller, 2018-09-21:
@weixingsun: Because I verified that this is working as expected:

[image]

weixingsun, 2018-09-23:
@mikeller Appreciate for your time for verifying this issue, it must be my problem.
I am using the same MSP panel, it's wired, mine doesn't work, 
I reviewed all your comments, you mentioned TX should be connected, my TX is a modified module
So it may be the root cause, what kind of TX are you using ?

mikeller, 2018-09-24:
@weixingsun: This was tested on a bare MATEKF722 board using the MSP RX.

weixingsun, 2018-09-27:
@mikeller I bought a MATEKF722, and my old OmnibusF4v3, still I cannot make both of them work as you shown
I don't know what is the difference, please help when you got chance to see this, thanks.
[image]
[image]

weixingsun, 2018-10-13:
@mikeller do you know what else to check or any possibility for my problem? Thanks