betaflight/betaflight-configurator #1204: 'Barometer' in sensors tab is not barometer, but estimated altitude

jflyper, 2018-09-23:
It has been estimated altitude for a long long time.

McGiverGim, 2018-09-23:
It only prints the result of the msp command. Is better to change the firmware or simply change the text in the configurator by ALT?

jflyper, 2018-09-24:
I have a mixed thoughts about this. The MSP itself is `MSP_ALTITUDE` and the current semantics is correct, and as I originally realized this while I was looking at an alternative  altitude code mikeller wrote, I preferred to be altitude. However, as the purpose of the sensors tab is to show how each sensor is functioning, straight barometer readout should be correct...

mikeller, 2018-09-24:
If anything we should have both - 'barometer' readout to check the sensor output, 'altitude' readout to check that the altitude estimation (with all sensors) is working correctly, probably _not_ in the 'Sensors' tab.

timman2er, 2018-10-07:
If anything, why is it's value limited to be between -1 and 1 right now? The barometer could read way different values on the bench.

jflyper, 2018-10-07:
@timman2er Graphs in sensors tab are auto scaling.

closed 2018-11-13 by stale[bot]