betaflight/betaflight-configurator #1206: OSD does not show settings

flash-v, 2018-09-29:
[image]

flash-v, 2018-09-29:
Because I can't use on the last stable release for Mac OS (not run) and there is a need to compile firmware by myself from last code.

jflyper, 2018-09-30:
I have no trouble running 10.4.0 on OSX 10.13.6.
What kind of "not run" symptom do you have?

flash-v, 2018-09-30:
[image]

The program is corrupt and can not be opened. Move the program to the trash.

jflyper, 2018-09-30:
Try downloading a fresh dmg if you haven't done so.

flash-v, 2018-09-30:
Downloaded today, an hour ago.

jflyper, 2018-09-30:
I think it has something to do with the permissions of files in the package.

What is your osx revision?

flash-v, 2018-09-30:
10.13.6

jflyper, 2018-09-30:
How is your osx Security & Privacy "Allow apps downloaded from:" configured?

flash-v, 2018-09-30:
"App store and from established developers

flash-v, 2018-10-01:
Yes, you were right. Now everything works. Thank you so much.

mikeller, 2018-10-01:
@flash-v: If the issue is resolved, can you please close it?

closed 2018-10-01 by flash-v