betaflight/betaflight-configurator #1209: Google MAPs load issue

fotoamg, 2018-10-04:
[image]
See map message...

mikeller, 2018-10-07:
Tested, Google maps works fine for me on the GPS tab. Could it be that you have a personal firewall or similar that is blocking configurator from accessing Google maps?

fotoamg, 2018-10-07:
Hi today it fixed itself, I have no firewall and antivirus sw. I changednothing since tha error occured first, could be a temporary issue as works now:
[image]

closed 2018-10-07 by fotoamg

mikeller, 2018-10-07:
Could have been an outage of Google (yes, even they have that).