betaflight/betaflight-configurator #1212: can not write new font to osd

robschii, 2018-10-08:
when i want to flash a new font (digital) after this the configurator close. 
[image]

omnibis f4  (Flame F4)

robschii, 2018-10-08:
bf 3.5.1

mikeller, 2018-10-08:
It is normal that the flight controller reboots and the connection drops after the font has been flashed. Do you see the progress bar during the flashing process? Have you checked if the new font has been uploaded?

robschii, 2018-10-09:
The progress bar runs to 100%
Sometimes its reboot sometimes not.  But the font did not change.  The flame f4 is a new one because my old flame f4 which I use more than 1 year lost the osd completely. This was the reason to change the fc

mikeller, 2018-10-09:
Could it be that the flame is a HGLRC product?

They are well known for crappy product quality, and in particular for using cheap and unreliable clones of the MAX7456 chip that is used for the OSD. Looks like the one on your second board is not fully functional - time to get a refund.

robschii, 2018-10-09:
Okay, I was the opinion that hglrc was good quality...   

This flame I had bought with the other one long time ago... 😢

timman2er, 2018-10-09:
should be closed then.

skulldrones, 2018-10-23:
I had the same problem to change font in FuryF3 (Diatone): the font has not updated. Work well in BFC 10.3.1, but don't work in BFC 10.4.0.

closed 2018-11-29 by stale[bot]