betaflight/betaflight-configurator #1218: Licensed resource

Slavvkko, 2018-10-22:
I should have asked earlier.
Recently I released configurator for android on google play https://play.google.com/store/apps/details?id=com.hordiienko.powerrate. In my project I use you mixer images and 3d models. I want to ask you: Can I use it? I have no plan for sell this app. Maybe I will release source code in feature, but don't know when.

McGiverGim, 2018-10-22:
Maybe others can correct me, I'm not an expert, but Betaflight uses GPL v3, that implies that if you use some code or any other resource, you must publish your code.

mikeller, 2018-10-22:
@Slavvkko: @McGiverGim is correct. Betaflight configurator is GPLv3, which means that if you use any part of it, or create a derivative work, your project has to be released under the GPLv3 as well. You have to release the source code no later than when you release the complete app to users.

closed 2018-11-28 by stale[bot]