betaflight/betaflight-configurator #1219: 10.4.0 - Change Betaflight logo in OSD don't work.

skulldrones, 2018-10-23:
When trying to replace the Betaflight logo, in:

 OSD tab > Font Manager > Select Custom Image, the image is selected and presented in the preview. 

After execute the "Upload Font" and the "Updating" ended, the new image is not displayed on the OSD after reboot.

PS.: The same image file and the same process are tested in 10.3.1 and worked properly.

mikeller, 2018-10-23:
What flight controller are you using?

skulldrones, 2018-10-23:
I had tested in 2 differents flight controllers: Fury F3 (Diatone) and AIOF3_BRUSHED

mikeller, 2018-10-23:
Ok. AIOF3_BRUSHED is not a target that is supported by Betaflight, so I can't help you there. I just verified that custom image / font flashing works in 10.4.0, but there's a slim chance that the OSD chip on your Fury F3 is busted and won't allow updating. Ask your supplier for a warranty exchange.

skulldrones, 2018-10-23:
(1) AIOF3_BRUSHED is not a target that is supported by Betaflight - OK, but Betaflight change logo in 10.3.1.
(2) FuryF3 has changed image in 10.3.1 but not in 10.4.0. I think this is not a hardware problem.

wind0r, 2018-10-23:
Flashing is also working for me with multiply Flight Controllers.
This is a hardware or user issue. Maybe you also should power the quad with a Lipo.