betaflight/betaflight-configurator #1223: PID tuning tab - error in drawing code - the rate lines and the simulated model are gone

bujarek, 2018-11-01:
There is an error in the code in the PID tuning tab. Whenever I enter this particular tab, the rates lines are not drawn and the drone model (in the Rates Preview window) is gone.
Also, if now (after entering the "PID tuning") I go back to the Setup tab, then the drone model (in the calibration window) is gone as well. The same applies to the drone model in the Receiver tab. It is reproducible, every time - once I open the PID tuning - no more drone models in every other part of the configutator.
I have checked the version 10.1 and that one worked correctly. The reinstalled 10.4 and the problem is back.

bujarek, 2018-11-01:
Updated the title - tested all 10.1.0, 10.2.0, 10.3.1 and 10.4.0 versions. 
Something has gone wrong after 10.2.0 - this is the last version that displays the rate lines and does not cause the 3d drone model to disappear.

McGiverGim, 2018-11-01:
Maybe something with the chrome version packaged and your OS? What OS are you using?

bujarek, 2018-11-01:
[image]
[image]

McGiverGim, 2018-11-01:
What I can say is that it is working for me, so is some kind of incompatibility with your system. Let's see if others appear with the same problem, to see if we can find something similar.

bujarek, 2018-11-01:
Can I help you debug 10.4.0 or whatever is currently in the repository?

This is very annoying because only this particular tab (PID tunning) is causing the issue. I can see the 3d drone model in all other tabs until I dare to open PID tuning :) 
Once I open the PID tunning tab I need to close the configurator application completely to see the drone model again. In other words: just disconnecting and reconnecting does not fix the view.

McGiverGim, 2018-11-01:
Maybe you can execute a debug version of the configurator, and take a look to the developers console to see if something is wrong. I'm out of home this week so I can't build it for you now.

bujarek, 2018-11-01:
Ok, i have setup node.js and the rest of the tools, and downloaded the lastest code from the repo. Compiled everything and am able to start npm start and the application starts, works, and I can see the same issue. How do I go from there? Where do I find the developers console? I'm a C programmer, never used node.js before.

McGiverGim, 2018-11-02:
Execute it with "gulp debug", and the console is in the mouse right button.

mikeller, 2018-11-03:
@bujarek:

There is nothing wrong with this function:

```
RateCurve.prototype.getMaxAngularVel = function (rate, rcRate, rcExpo, superExpoActive, deadband) {
    var maxAngularVel;
    if (!this.useLegacyCurve) {
        maxAngularVel = this.rcCommandRawToDegreesPerSecond(maxRc, rate, rcRate, rcExpo, superExpoActive, deadband);
    }

    return maxAngularVel;
};
```

In JavaScript, every variable is automatically initialised to `undefined`. And since the code calling `getMaxAngularVel()` is only ever executed when `!this.useLegacyCurve` is true, returning undefined in this case is fine.

`RateCurve.maxAngularVel` is used in `pid_tuning.js`:

```
mikeller@pc:~/git/betaflight-configurator$ grep -r rateCurve.maxAngularVel src
 
src/js/tabs/pid_tuning.js:    if (!self.rateCurve.useLegacyCurve && self.rateCurve.maxAngularVel) {
src/js/tabs/pid_tuning.js:                maxAngularVel       = self.rateCurve.maxAngularVel,
```

Unfortunately I suspect you will have to debug / fix this yourself - I cannot reproduce the behaviour you are describing either.

bujarek, 2018-11-03:
@mikeller:
Yes, I understand that if you check the !self.rateCurve.useLegacyCurve then your code is safe.
However, if someone adds a call without that check in place - they will loose time trying to find out what is going on. 
So, it would be nice to change the function to:
`RateCurve.prototype.getMaxAngularVel = function (rate, rcRate, rcExpo, superExpoActive, deadband) {
    var maxAngularVel;
    if (!this.useLegacyCurve) {
        maxAngularVel = this.rcCommandRawToDegreesPerSecond(maxRc, rate, rcRate, rcExpo, superExpoActive, deadband);
    }
    else
    {
        console.log('ERROR: add a check in your code for !useLegacyCurve before calling the RateCurve.getMaxAngularVel() - returning undefined value now!');
    }

    return maxAngularVel;
};`

bujarek, 2018-11-03:
@mikeller:
Could you help me understand this code, since this is related to the original issue (the above function is not related, just something I noticed):
```
RateCurve.prototype.setMaxAngularVel = function (value) {
    this.maxAngularVel = Math.ceil(value/200) * 200;
    return this.maxAngularVel;
};

```

mikeller, 2018-11-03:
@bujarek: See https://github.com/betaflight/betaflight-configurator/issues/1223#issuecomment-435554729:

> RateCurve.maxAngularVel is used in pid_tuning.js

McGiverGim, 2018-11-03:
Which values are in the variables of the method when it fails?

bujarek, 2018-11-03:
The values are in order:

axisLabel: 1000 deg/s 
x: 629.16 
y: 61.199999999999996

These look perfectly valid, right?

bujarek, 2018-11-03:
During this intesive tests I have found two minor issues, I guess.

1) Whenever one of the RC rates=0.00, then the Max Vel column displays "NaN" and all three rate lines disappear. 

2) If I start dragging by the right window border, making the client window smaller (narrower) there is a moment when the area for the rates becomes bigger than its parent frame (see the image).
[image]

mikeller, 2018-11-03:
@bujarek: x and y are in pixels, so they should be integers. Does `context.fillText('1000 deg/s', 629, 61)` work for you?

Also, as you said, one of the problems is that Windows 7 is not fully supported by Chrome, and implicitly bye the NW.js toolkit that is built from the Chrome sources. Betaflight configurator normally updates the NW.js version used for every release, so the fact that this is working for you with older versions but not with the latest ones could be based on the fact that the NW.js used has been updated to a version that does not support Windows 7 any more.

There definitely is a problem when RC rates are set to 0 - this should never be possible because the craft will be unfliable in this configuration.

bujarek, 2018-11-04:
@mikeller : actually x and y are vars, so they can hold real numbers as far as I can tell. This is what is output in the console:
axisLabel:1000 deg/s x: 956.4800000000001 y: 61.199999999999996
just with this simple code: `console.log('axisLabel:' + axisLabel + ' x: ' + x + ' y: ' + y);`

Anyway, the `context.fillText('1000 deg/s', 629, 61);` does not work either.
I've been checking a lot of combinations and it seems it does not cause the issue if I type:
`context.fillText('A', 0, 0);` - but 'A' is not visible in the graph either, just that the 3d drone model is still displayed with that, so not a fix at all.

Does anyone use Windows 7 here?

bujarek, 2018-11-04:
ok, one hour later I found that my code using my newly added (tmp) canvas works just fine and displays whatever I want. hmmm... the plot begins... :) Finally, I found out that the problem the fillText() crashes silent was that the canvas size cannot exceed 256x256. I've tried 256x257 and 257x256 and both fail, anything below 64kB works fine. 

So, the question to the maintainers: if I prepare a fix that will decrase these two canvas size to 256x256, would that be accepted?

bujarek, 2018-11-04:
It seems the issue is known, and some kind of a workaround is available:
https://www.tutorialspoint.com/HTML5-Canvas-and-z-index-issue-in-Google-Chrome

mikeller, 2018-11-04:
Good find, @bujarek! Can you please open a pull request to fix this?

McGiverGim, 2019-05-22:
@bujarek have you tested the latest Configurator 10.5.1 to see if the problem persists? Some fixes to the way the rate of the 3d quad were merged. I don't think it is related with your problem, but perharps...

haslinghuis, 2021-06-05:
Closing as inactive for more than one year

closed 2021-06-05 by haslinghuis