betaflight/betaflight-configurator #1225: Saving on failsafe tab disables GPS-Rescue Failsafe

Footr, 2018-11-06:
I activated GPS-Rescue failsafe via the CLI.  I then went to the failsafe tab to change pulse timings.  After I saved that change GPS-Rescue was replaced with "Land" when I checked the diff.

McGiverGim, 2018-11-06:
The save in the failsafe has only the old two options. Until the GPS rescue is added to this tab this is an expected behavior because the value is saved in the same parameter. I will look if it has been added to MSP, if added then this tab can be updated.

mikeller, 2018-11-06:
@McGiverGim: Even if we do not yet have support for the parameters for GPS rescue, we could always just add support for selecting the mode itself (to avoid resetting it when the user saves the tab), and add a note that the parameters have to be set in CLI.

McGiverGim, 2018-11-06:
I will look into it next week if nobody does it before

closed 2018-11-14 by mikeller