betaflight/betaflight-configurator #1239: Windows 10 cannot build betaflight configurator 10.4.0/10.4.1 for win32/win64. Found solution.

melenefoti, 2018-11-27:
My dear friends, i was trying to compile 10.4.0 / 10.4.1 betaflight configurator on windows 10, but while executing "gupl release --win32 --win64" the compile was crashing on "getChangesetId".
So, i commented the line 64 "gulp.task('get-changeset-id', getChangesetId);"
and changed the line 66: var distBuild = gulp.series(dist_src, dist_locale, dist_libraries, dist_resources, getChangesetId);
with this var distBuild = gulp.series(dist_src, dist_locale, dist_libraries, dist_resources);

removed "getChangesetId" element.

now it is compiling properly.

I attach a screenshot of the error
[image]
.

mikeller, 2018-11-27:
@melenefoti: Looks like the git repository you are trying to build from is broken: 

[image]

Try cloning the repository into a new directory, and building from there.

melenefoti, 2018-11-27:
thank very much, you are right. fixed the problem. i am closing the issue.

closed 2018-11-27 by melenefoti