betaflight/betaflight-configurator #1242: Inav Configurator- fail to close serial port

Cyremiaepf, 2018-12-02:
Hello,

I'm actually working on a MATEK f405-STD and i have an issue when i plug it on inav. Every time i try to save a new preset and i need to reboot, there is a failure during the process:

 EEPROM saved

2018-12-02 @ 15:51:05 -- Device - Rebooting

2018-12-02 @ 15:51:06 -- Unrecoverable failure of serial connection, disconnecting...

2018-12-02 @ 15:51:06 -- Failed to close serial port

I used zadig and Impulse RC driver to set the drivers.

Any ideas of how to fix it ?

Thanks

etracer65, 2018-12-02:
This is the github for the Betaflight Configurator - not iNav.

I think you want here:
https://github.com/iNavFlight/inav-configurator

closed 2018-12-02 by etracer65

Cyremiaepf, 2018-12-02:
Sorry, thanks !