betaflight/betaflight-configurator #1244: Layout issue with small window

hydra, 2018-12-03:
I use two monitors in portrait mode and noticed that there's a layout issue as per following screenshot.

[image]

Using `master` from today.

hydra, 2018-12-03:
resolution of my portrait monitors are 1920x1080p, so width is 1080 pixels WIDE (not high).  Screenshot above is with a width of 1066 pixels.

klutvott123, 2018-12-05:
It's caused by the addition of the "update firmware" button i think. Before that was added, the minimum width was restricted just enough to keep that from happening. When we're talking about layout issues, try resizing the PID tuning page. 😃