betaflight/betaflight-configurator #1245: Dataflash download fails with exception in console.

hydra, 2018-12-03:
Exception occurs when trying to download dataflash, as below.

```
Dataflash dump file path: D:\My Documents\dev\embedded\betaflight\BTFL_BLACKBOX_LOG_20181203_100522.BBL
main.html#:1 Error in event handler for serial.onReceive: RangeError: Invalid DataView length undefined
    at chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/msp/MSPHelper.js:1786:45
    at MspHelper.process_data (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/msp/MSPHelper.js:1280:31)
    at chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/msp.js:139:13
    at Object.notify (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/msp.js:138:24)
    at Object.read (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/msp.js:126:26)
    at read_serial (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/serial_backend.js:458:13)
```
Screenshot:

[image]

mikeller, 2018-12-03:
FCT1 does not seem to be a flight controller supported by Betaflight. Can you reproduce this with a supported flight controller?

hydra, 2018-12-03:
neucleo f722

mikeller, 2018-12-03:
The NUCLEOF722 does not support data flash, that's why it's not working.

hydra, 2018-12-03:
It does if you enable it like you would for any new target.

mikeller, 2018-12-03:
Can you please supply a reproduction for a target that actually exists, or leave this closed?

hydra, 2018-12-03:
not currently no, however.

Given:

`function onChunkRead(chunkAddress, chunkDataView, bytesCompressed)`

and:

`MspHelper.prototype.dataflashRead = function(address, blockSize, onDataCallback) {`

and:

`onDataCallback(address, new DataView(response.data.buffer, response.data.byteOffset + headerSize, dataSize));`

you can see the arguments for the callback are not correct hence the exception I posted.

line above, in MSPHelper.js should be.

`onDataCallback(address, new DataView(response.data.buffer, response.data.byteOffset + headerSize), dataSize);

like I said, there's a bug.

mikeller, 2018-12-03:
If you've got a fix, then why do you open an issue for a board that does not exist in Betaflight and ask for others to go on wild goose chases, instead of just opening a pull request with a fix?

closed 2018-12-03 by mikeller

hydra, 2018-12-03:
Here's a screenshot with the PR applied.

[image]