betaflight/betaflight-configurator #1250: Flasher fails to detect DFU after sending "R"

jflyper, 2018-12-07:
The board goes into DFU, "DFU" is displayed in the port list pull down, but flasher is wedged.

```
Loading file from: ~/Dropbox/repo/cleanflight/obj/betaflight_4.0.0_NUCLEOH743.hex
firmware_flasher.js:382 File loaded
js/workers/hex_parser.js:106 HEX_PARSER - File parsed in: 0.0680 seconds
serial.js:138 SERIAL: Connection opened with ID: 2, Baud: 115200
stm32.js:90 Sending ascii "R" to reboot
serial.js:266 SERIAL: Connection with ID: 2 closed, Sent: 1 bytes, Received: 0 bytes
port_handler.js:30 PortHandler - Removed: /dev/tty.usbmodem2420,/dev/cu.usbmodem2420
main.html#progressbar:1 Error in response to usb.getDevices: Error: Invocation of form usb.getDevices(undefined, function) doesn't match definition usb.getDevices(object options, function callback)
    at STM32DFU_protocol.connect (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/protocols/stm32usbdfu.js:92:16)
    at chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/protocols/stm32.js:109:50
    at Object.callback (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/port_handler.js:164:22)
    at Object.PortHandler.check_usb_devices (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/port_handler.js:150:16)
    at chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/protocols/stm32.js:107:45
```

mikeller, 2018-12-07:
@jflyper: What version / changeset of the configurator is this? The stack trace looks like the one that caused the issue fixed in #1241.

jflyper, 2018-12-08:
@mikeller That fixed it! Good find on #1241. 👍

closed 2018-12-08 by jflyper