betaflight/betaflight-configurator #1253: deb package installs files with user permissions

christianrauch, 2018-12-09:
The configurator debian package is installed in `/opt/betaflight/betaflight-configurator`. Usually, files installed to a system directory have write permissions for the `root` user.
However, the betaflight-configurator debian package is installed such that the first user (id 1000) owns the folder `/opt/betaflight` and also has write acces.

The `/opt/betaflight` folder should be own bei either the `root` user or a dedicated `betaflight` user, but definitely not by user id 1000.

mikeller, 2018-12-09:
Oops.

binlan, 2019-03-23:
this happend because the packages are build as a user with uid 1000:

```
marc@sr-6:~/betaflight-configurator$ tar -tvf data.tar.xz |head -n4
drwxr-xr-x mikeller/mikeller 0 2018-08-13 12:54 ./
drwxr-xr-x mikeller/mikeller 0 2018-08-13 12:54 ./opt/
drwxr-xr-x mikeller/mikeller 0 2018-08-13 12:54 ./opt/betaflight/
drwxr-xr-x mikeller/mikeller 0 2018-08-13 12:54 ./opt/betaflight/betaflight-configurator/
marc@sr-6:~/betaflight-configurator$ tar --numeric-owner -tvf data.tar.xz |head -n4
drwxr-xr-x 1000/1000         0 2018-08-13 12:54 ./
drwxr-xr-x 1000/1000         0 2018-08-13 12:54 ./opt/
drwxr-xr-x 1000/1000         0 2018-08-13 12:54 ./opt/betaflight/
drwxr-xr-x 1000/1000         0 2018-08-13 12:54 ./opt/betaflight/betaflight-configurator/
```

packages are build by the module gulp-debian:
```
marc@sr-6:~/betaflight-configurator$ grep gulp-debian gulpfile.js 
const deb = require('gulp-debian');
```
Unfortunately this package calls dpkg-deb without the option `--root-owner-group`.
```
marc@sr-6:~/betaflight-configurator$ grep dpkg-deb node_modules/gulp-debian/index.js 
        _exec(`dpkg-deb --build ${pkg._out}/${pkg.package}_${pkg.version}_${pkg.architecture}`,
```
A workaround could be to run the build (gulp release) aus root or patching the line in the module after download. I think there is no way to set this switch as a param.

mikeller, 2019-03-23:
Thanks @binlan, good find! #1327 and betaflight/blackbox-log-viewer#288 should address this.

McGiverGim, 2019-04-03:
I close this, in theory it is fixed.

closed 2019-04-03 by McGiverGim