betaflight/betaflight-configurator #1255: Can't add in chrome nor compile with npm

fgiudice98, 2018-12-14:
Adding in chrome give first a missing icon error, second a missing locales error and then the missing eventPage.js.
I then installed node.js and npm, but I can't start npm as it gives me
`23 error If you do, this is most likely a problem with the betaflight-configurator package,
23 error not with npm itself.
23 error Tell the author that this fails on your system`

I use an ubuntu subsystem on win10 and chrome latest on the same system.

McGiverGim, 2018-12-14:
If you are under windows 10, is easier to install npm for Windows and not use the ubuntu subsystem.

mikeller, 2018-12-15:
@fgiudice98: Have you tried googling the error message? When I do so, I get https://github.com/gulpjs/gulp/issues/2246 as one of the top results, complete with instructions on how to fix it.

fgiudice98, 2018-12-15:
@mikeller Yeah, I was just writing that. I have to install a lot of dependencies, so my fault, sorry for that.
For chrome? I'm missing something? I tried many things (changing files) but nothing works.

closed 2018-12-25 by fgiudice98