betaflight/betaflight-configurator #1257: rebooting freezes - needs pressing 'disconnect' twice, then 'connect'

RipperDrone, 2018-12-22:
Tries 12/19 Nightly of BFC with BF 4.0 several Nightlies from last week: Whenever I save + reboot from any BFC tab or CLI, BFC stays in connected state despite the FC rebooting. BFC needs double press of 'disconnect', then 'connect' to re-connect... looks like it's a matter of bad port open/close timings?

FCs were OBF4, OBF4SD and SpeedyBeeF4...

mikeller, 2018-12-22:
@RipperDrone: Is this actually a bug? I.e. do you **have** to disconnect / reconnect, or will configurator just continue to work as designed after 'Save and Reboot' and the automatic reconnect? Configurator was designed to automatically reconnect after a  'Save and Reboot', and this works well for targets that use a dedicated USB to serial converter. For targets with USB VCP this works if most peripherals are disabled, since the peripherals are initialised before USB is configured, and this often leads to an USB disconnect that causes the automatic reconnect to time out.

RipperDrone, 2018-12-22:
@mikeller : I haven't configured automatic reconnect in BFC. This is what happens in detail:

1. Make the FC reboot e.g. by issuing a CLI save 'command' --> FC reboots (PC sounds USB port disconnect/reconnect), but BFC is stuck in connected state: No reaction when trying to change tabs, pressing buttons, anything
2. I have to press 'disconnect' in BFC TWICE, then the 'connect' button shows active again, and I can press it to finally connect

I do not have to physically unplug/plug in the USB cable. It#s just the port open/close timing seems glitchy so BFC doesn't detect the port close/port reopen (or in general the FC's reboot) properly...

RipperDrone, 2018-12-23:
It worked flawlessly with same FCs and previous BFC version 10.4, so something in the port handling / timings seems to have changed.

RipperDrone, 2018-12-23:
I reloaded BFC 12/19 Nightly over Chrome/Developer mode/Unpack app (extension). Now suddenly it reconnects automatically after reboot! Maybe updating to Nightlies by extracting the .zips into the current (Chrome) app directory was not the right way to do it. Seems to have been my own fault - closing this issue ...

closed 2018-12-23 by RipperDrone