betaflight/betaflight-configurator #1264: `parity_error` after issuing `save` in cli.   Connect button breaks.

hydra, 2019-01-05:
Steps to repeat.

1. start configurator in chrome.
2. connect, goto CLI tab, type in issue `save`
3. observe console error as below.
```
{connectionId: 6, error: "parity_error"}
connectionId: 6
error: "parity_error"
__proto__: Object
```
4. observe that 'Disconnect' button causes the following error to display. 
```2019-01-05 @ 14:04:13 -- Please use the Firmware Flasher to access DFU devices```
Note that the device had rebooted into DFU mode instead of a regular reboot, unsure if this is related at this point as I'm working on firmware when this occurred.
5. observe that unplugging USB cable still did not cause the UI to recover.

Suggest adding handling for 'parity_error' into `serial_backend.js` that handles it as per `device_lost` and `break`

mikeller, 2019-01-07:
Tried to reproduce this with:
- Betaflight configurator 10.4.1;
- OMNIBUSF4, firmware 3.5.4;
- linux.

Unable to reproduce this. Some details on your configuration will make this reproducible / fixable.

hydra, 2019-01-07:
Windows, Chrome, latest commit was 07f28e44. BF from latest master.

You should be able to replicate the issue if you connect the configurator to an FC, then connect another FC in DFU mode. Then use CLI `exit` on the connected FC.

For me, on windows, I find the configurator often fails to disconnect when a VCP FC reboots.  When this happens (which for me is 99.9% of the time right now) and you have another device (or for me, same same device) in DFU mode, the configurator's logic will be wrong because it does 'DFU' connection stuff first, before dealing with the actual logic of  disconnecting when a disconnect caused by serial error (`parity_error`) has already happened.

see

https://github.com/betaflight/betaflight-configurator/blob/master/src/js/serial_backend.js#L38-L56

The value of `selected_port` can become `DFU` when you're actually connected, after a serial connection error has occurred, so you can't disconnect.

hydra, 2019-01-08:
@mikeller I think you're mis-understanding when this occurrs.

'parity_error' for me occurs on windows with the FC board I'm developing.  However, any un-handled serial error will cause the connect button to break: 
* if the device that causes the serial error reboots into DFU mode.
* if there is another device connected, already in DFU mode when the serial error occurrs.

mikeller, 2019-01-08:
@hydra: Fair enough, as long as we have a reproduction that is available to any developer there is a chance that somebody will fix this.

The easiest will probably be to add a catch-all error handler.

hydra, 2019-01-11:
on an SPRAINGF7DUAL today I also got this error when `save`ing at the CLI.  Also needs handling.

```
{connectionId: 19, error: "frame_error"}
connectionId: 19
error: "frame_error"
__proto__: Object
```

Docteh, 2019-07-03:
`console.error(info);` 

This isn't happening whenever I want it to :(
Instead of a nice json object I get this to play with in the console: 
[image]

McGiverGim, 2019-07-04:
It shows the complete object, including methods, definition, etc. What does the `info.toString()` method print?

Docteh, 2019-07-04:
Turns out the complete object details are available there if you open the console before having a serial error. I was consistently opening the console after having a serial error. 

Console open before typing save on CLI, works first time:
[image]

I get this if I `save` twice in the CLI, and then open the inspector/console window
[image]

PR #1522 (2019-07-08): add a default handler for serial errors (disconnect)
Also cleaned up some tabs and extra spaces in the file. The 'disconnected' case has been moved, to respect its current status of doing nothing.

This might be a solution to fix #1264 I guess if it doesn't @hydra can just reopen it?
closed 2019-07-08 by mikeller