betaflight/betaflight-configurator #1270: [Feature Request] support mode logic (AND/OR) in GUI

howels, 2019-01-11:
https://github.com/betaflight/betaflight/pull/4722 introduced ability to enable modes using the boolean AND of ranges; for instance "set camera OSD mode if copter AUX1 <1600 AND AUX3>1600" instead of using the default "OR" logic.

The CLI syntax changes from:
```CLI_COMMAND_DEF("aux", "configure modes", NULL, cliAux)```

To: 

```CLI_COMMAND_DEF("aux", "configure modes", "<index> <mode> <aux> <start> <end> <logic>", cliAux),```

However the GUI doesn't recognise this. It would make it much easier to configure multiple modes if the AND/OR logic was implemented in the GUI

McGiverGim, 2019-01-12:
Is there an MSP implementation of this? I only see CLI changes in the PR

mikeller, 2019-01-12:
@McGiverGim: I don't think MSP support for this has been added yet - that is part of the work that should be done.

dthubereng, 2019-01-13:
> @dthubereng: You wouldn't have time to add UI support in configurator for this? 😉

That would require me to learn Java, correct? Or is it Javascript? Either of those are pretty non-existent in my repertoire at the moment. I'm not averse to diving in though.

McGiverGim, 2019-01-14:
This needs C first adding the feature to MSP in the firmware, and later Javascript in the configurator.

mikeller, 2019-01-14:
@dthubereng: As @McGiverGim says, in a first stage the new parameters will have to be added to the MSP protocol in the firmware (see `src/main/interface/msp.c`). And then the client side will have to be done, which is JavaScript.
Happy to support you if you want to have a go at it.

dthubereng, 2019-01-14:
I will take a look at it.

dthubereng, 2019-01-17:
Firmware is done (no PR yet). Working through the configurator changes now in order to test both. Question: any recommendations on dev environments for running compiled Javascript? I've been using VSCode, which is great on both my mac and windows machines for writing firmware, but I don't really use it for debugging (no debug hardware other than an actual FC). @mikeller?

McGiverGim, 2019-01-17:
The debug is done from chrome, if you start in debug mode `gulp debug` you can use the developers console (right click with the mouse).

mikeller, 2019-01-17:
@dthubereng: I mostly use the debug console as well, but I've heard good things about  debugging NW.js (which is the framework the configurator is built in) from VS Code. Try https://marketplace.visualstudio.com/items?itemName=ruakr.vsc-nwjs

dthubereng, 2019-02-05:
I'm pretty much done with this, but need to check on a few things before I submit the PR. I've never done any html, css, or js/jquery so this has been quite an experience and I will gladly accept any critique. I also added the configuration for the linkedTo option but didn't realize the two (logic and linked) were somewhat unrelated and should have been two separate PRs. So, the change will be simply adding an extra MSP message and the supporting GUI. PR forthcoming...

PR #1287 (2019-03-02): Add modeLogic and mode link UI support
Fixes #1270 

Add MSP_MODE_RANGES_EXTRA to msp codes
Add MODE_RANGES_EXTRA function call
Add modeRangeExtra when reading mode settings
Add UI for modeLogic
Position and populate Logic selector
Some fixes for msp apiVersion checking
Add 'save' modeLogic
Add link button and translations
Add code to show/hide Link button
Adjust position of link and range buttons
Add structure code for Link button
Add linkedTo code
Add link div with delete button to html and css.
Add code to hide/show buttons when link is pressed.
Add code to add link select to range area.
Ad [... cut]
closed 2019-03-02 by mikeller