betaflight/betaflight-configurator #1276: New GUI tab for GPS Rescue mode

WohaibHasan, 2019-01-16:
i am new, and not sure if this already planed feature.
i suggest to have a new "Rescue mode" GUI tab in the betaflight-configurator that can be used to easily configure: 

1. pre-configured Track
2. GeoFence,  , 
3. back-to-home point (Return to Home)
4. Failsafe

mikeller, 2019-01-16:
The 'GPS rescue' feature was developed (the name might be a giveaway there) to serve as a means of last resort to bring a craft back to its origin if the RC connection is lost. Features 1, 2, 3 that you list do not relate to this functionality at all. Also, these features are far outside of the intended use of Betaflight (racing / freestyle flying with high performance craft), so there is little to no interest in implementing and maintaining these in the developer community.
If you are looking for these features, you should have a look at [iNav](https://github.com/iNavFlight/inav/wiki).

WohaibHasan, 2019-01-17:
thanks **mikeller** for the clarification and for your efforts in making the betaflight :)

closed 2019-01-17 by WohaibHasan

pkarc, 2019-02-12:
I'm able to do and add the standard gps_rescue co figuration UI to betaflight configurator(https://github.com/betaflight/betaflight/wiki/GPS-rescue-mode), nothing fancy. Also add failsafe_procedure specially because it got reset every time you change something at the failsafe tab.

McGiverGim, 2019-02-12:
This was added some time ago... https://github.com/betaflight/betaflight-configurator/pull/1228