betaflight/betaflight-configurator #1279: USB 3.0 friendly?

SRTDRE76, 2019-01-19:
Why isn’t BetaFlight connecting and asking for usb 2.0 due to incompatibility with usb 3.0 and detection of FCs. And why can’t the required drivers be installed by the configurator setup.executable and have them recognized by windows. I mean no tablets or PCs are sold with usb 2.0 any longer.

mikeller, 2019-01-19:
@SRTDRE76:

> Why isn’t BetaFlight connecting and asking for usb 2.0 due to incompatibility with usb 3.0 and detection of FCs.

I do not quite understand this, can you elaborate a bit?

> And why can’t the required drivers be installed by the configurator setup.executable and have them recognized by windows.

Good point, feel free to open a pull request adding USB drivers and integrating them in the installer.

jflyper, 2019-01-25:
> Why isn’t BetaFlight connecting and asking for usb 2.0 due to incompatibility with usb 3.0 and detection of FCs.

I don't understand this. FCs are limited to USB2.0, so users have to use USB3.0 to USB2.0 hub (or dongle or adapter or whatever form factor it comes in). With the hub, the USB2.0 devices are transparently handled by USB2.0 drivers. This is, of course, transparent to the configurator, too.

In summary, the configurator is not asking for USB2.0.

closed 2019-03-03 by stale[bot]