betaflight/betaflight-configurator #1293: Receiver Tab cleanflight/betaflight not working

Tom12602, 2019-02-10:
I built an fpv drone with a HGLRC XJB F440 F4 flight controller and a Turnigy ia6c receiver controller by a turnigy evolution transmitter. After soldiering everything together I plugged it into betaflight and found that my receiver wasnt showing any input on the receiver tab. I have replaced the receiver 3 times, flight controller 2 times, transmitter 2 times, changed from betaflight to cleanflight and back, reflashed 2 times, changed a bunch of settings and yes before someone asks i remembered to plug it into the battery. Ill attatch a few images, any help is appriciated.

Tom12602, 2019-02-10:
http://imgur.com/e0f7eOJ
http://imgur.com/VM3vHNr
http://imgur.com/VM3vHNr
My Drone. Ill send pictures of my betaflight config

Tom12602, 2019-02-10:
https://photos.app.goo.gl/zVXyCk8agG1PQHgc7
My Betaflight config

mikeller, 2019-02-10:
@Tom12602: This is not the right place to get help with configuration issues.

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

closed 2019-02-10 by mikeller