betaflight/betaflight-configurator #1299: Manual installation - Error with script js/eventPage.js

bungyguy, 2019-02-17:
Hi Guys

I wanted to use a feature for SPI_FRSKY_LBT which is not released yet in the BF Configurator exe for windows 10.4.1 and in chrome store.
So I tried to install the extension manually using the documentation: download master from github as zip, unpack zip, goto chrome extensions, enable dev mode, point to extracted BFC folder (betaflight-configurator-master) and install.

When I do this I get the error message: (translated)
Background script "js/eventPage.js" could not be loaded.
could not load manifest.

Please let me know how to fix this error or provide a working link with a nightly build of BF configuration.

Thanks in advance
Mike

bungyguy, 2019-02-17:
I am running widows 10 and Chrome 72.0.3626.109

Asizon, 2019-02-17:
Try wirh 
#set rx_spi_protocol= FRSKY_X_LBT

bungyguy, 2019-02-17:
I did that, but when I use the configuration page page in a released BFC version, the protokoll defaults to a different standart since SPI_FRSKY_LBT is not yet natively supported. So I wanted to use a BFC Version supporting this featue.
Can someone please send me a working BFC extension for Chrome from the current master please?

mikeller, 2019-02-18:
@bungyguy: The documentation is outdated (do you have a link to where you found it?). The betaflight-configurator repository cannot be run as a chrome web app directly any more, it needs some build steps (and locally installed build tools).
You can instead download the latest build from https://ci.betaflight.tech/job/BetaFlight_Configurator/ if you want to test new features, and don't mind potentially breaking your configuration.

bungyguy, 2019-02-18:
Hi Mikeller

Thanks a lot for your help! I will try your solution tonight and I am quite confident it will work.

I found the outdated documentation on the project page: https://github.com/betaflight/betaflight-configurator in section "Alternative way, Chrome app".
Maybe you can put the reference to https://ci.betaflight.tech/job/BetaFlight_Configurator/ onto the project page? This would be very useful.

Have a good day!
Mike

bungyguy, 2019-02-19:
It worked using the version from https://ci.betaflight.tech/job/BetaFlight_Configurator/

closed 2019-02-19 by bungyguy

mikeller, 2019-02-20:
Addressed in #1305.