betaflight/betaflight-configurator #1301: Betaflight OSD font doesnt change

fOmey, 2019-02-18:
OSD fon't doesn't seem to want to change..

Bar fills up to 100% when uploading a alternative font, FC reboots.. whack the goggles on and the font hasn't changed.. I've tried changing the font so the OSD preview updates and saving, that doesn't seem to make a difference.

I've tried selecting different fonts and it refuses to change from stock.

Diatone Mamba F405 Mini FC : https://www.diatoneusa.com/store/p574/MAMBA_F405_Mini_Betaflight_Flight_Controller_F25_25A_2_4S_DSHOT600_FPV_Racing_Brushless_ESC.html

```
# dump

# version
# Betaflight / FURYF4OSD (FY4O) 3.5.5 Jan 15 2019 / 09:56:01 (88921670b) MSP API: 1.40

board_name FURYF4OSD
manufacturer_id 

# name
name -

# resources
resource BEEPER 1 A08
resource MOTOR 1 A03
resource MOTOR 2 B00
resource MOTOR 3 B01
resource MOTOR 4 A02
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 C09
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
[... 700 lines cut ...]
set horizon_tilt_expert_mode = OFF
set abs_control_gain = 0
set abs_control_limit = 90
set abs_control_error_limit = 20

# rateprofile
rateprofile 0

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650
set throttle_limit_type = OFF
set throttle_limit_percent = 100
```

mikeller, 2019-02-18:
@fOmey: Font changing not working is often associated with cheap and unreliable video overlay chips being used on cheap clone boards. There is not much you can do except replace the board.

fOmey, 2019-02-18:
@mikeller: I was getting really consistent results by writing the font twice tho, which seems really odd..

Perhaps some sort of write verification would solve this.. between each block write, verify the data has actually taken hold, otherwise retry, maximum of 3 retrys before spitting an error?

mikeller, 2019-02-19:
@fOmey: So if you write the font, then after the flight controller has rebooted, write the font again straight away, it works all of the time?

If so, then adding repeats might be a way to beat those crappy clones into submission.

nerocide, 2019-03-28:
Same issue on DALRCF405, the only way to change the font is to plug a lipo then proceed with font upload.

Jetrell, 2019-04-14:
Had the same issue with the Diatone Mamba F405 mini.
It won't not load any fonts with USB powering the FC. 
But did load with the battery connected.

Jetrell, 2019-04-14:
Had the same issue with the Diatone Mamba F405 mini.
It won't load any fonts with USB powering the FC.
But did load with the battery connected.

mikeller, 2019-04-14:
@nerocide, @Jetrell: Looks like there is a clone version of the MAX7456 chip out there that is sensitive to the input voltage while flashing.

Jetrell, 2019-04-14:
@mikeller  This Mamba F405 mini came in the 2019 GT Rabbit 349.
You would think Diatone wouldn't be that tight :-1: 
Well at least I can still get it to function...better than not at all.

mikeller, 2019-04-14:
@Jetrell: Not sure if this is a case for 'tight' or 'not tight'. The part was discontinued by the original manufacturer, and I suspect now it's a matter of getting more or less functioning clones of dubious nature for manufacturers.

DPN217, 2019-04-14:
Had the same issue with the Diatone Mamba F405 FC and Betaflight 3.5.7.
I can load font with the battery connected.

nerocide, 2019-04-18:
@mikeller so this issue is not related to configurator problems, i would suggest to write a note on a wiki (yet unexistant) or something about this particular problem, stating known FC with chips problems.

mikeller, 2019-04-29:
@nerocide: Good idea. The wiki at https://github.com/betaflight/betaflight/wiki is editable for GitHub members, feel free to start this entry. I think the listing of affected parts / boards is best left to owners of this hardware anyway.

jflyper, 2019-04-29:
Users of MinimOSD and Micro-MinimOSD experienced font uploading problem with the original MAX chip (and the M328P chip), which was sensitive to the supply voltage and current budget when writing to font EEPROM. Although Fury F4 OSD seems to use AT7456E chip, I suspect the same problem.

mikeller, 2019-04-29:
@jflyper: Good point - current budget would also explain the reports from users that find they can only flash fonts with a battery plugged in.

nerocide, 2019-04-29:
Wiki page is created, and can be referenced in other wiki section 
 https://github.com/betaflight/betaflight/wiki/OSD-Font-upload-problem .

I guess we can now close this issue? @fOmey @Jetrell  @mikeller

mikeller, 2019-04-29:
@nerocide: Thanks.

@fOmey: Can you close this please?

closed 2019-05-01 by fOmey

naanne, 2021-01-24:
la solution : gestionnaire de police + charger le fichier de police = redemarre la carte avec le nouveau fichier de police ...... et voila !