betaflight/betaflight-configurator #1303: google maps problem in the GPS tab

laurentopia, 2019-02-18:
[image]

mikeller, 2019-02-21:
@laurentopia: Can you try again? Sometimes this is caused by an outage of Google.

McGiverGim, 2019-02-22:
The console message is clear: 

> You have exceeded your request quota for this API. See https://developers.google.com/maps/documentation/javascript/error-messages?utm_source=maps_js&utm_medium=degraded&utm_campaign=billing#api-key-and-billing-errors

The link in the capture points to the same link.

@mikeller who owns the API key? how many uses by day we have?

McGiverGim, 2019-02-22:
Maybe we are not using an API key at all? In theory, is a parameter passed to the call, but I don't see it:

https://github.com/betaflight/betaflight-configurator/blob/61392ca27a8952c370ec76d4f8a4727942e12eb2/src/js/tabs/map.js#L26-L31

mikeller, 2019-02-23:
From the look of it access to Google Maps through the JS API is no longer free:

https://developers.google.com/maps/documentation/javascript/usage-and-billing

I read about that somewhere a while ago, but did not connect the dots that Betaflight configurator will be affected.

In short, this means that we can no longer use Google Maps for the GPS map in configurator, as we don't want to cover the cost of our users using this tab.

Looking at iNav, they seem to have switched to using OpenStreetMap (or Bing maps, through openlayer, see iNavFlight/inav-configurator#595). If we want to re-enable map support for the GPS tab, we'll have to switch as well.

McGiverGim, 2019-02-23:
Looking at the internet, I see at some places that Google gives 200$ free at month that can be used in different places including maps. I'm not too sure about conditions.

If I get some time I can look at Open Street Map, is enough for our usage.

mikeller, 2019-02-23:
Yeah, I saw the 'free USD 200' offer too. But as far as I understand it this requires you to set up payment through a credit card, and once the USD 200 are used up, your card will be charged for the requests. So adding an API key to configurator would be playing with fire, as we'd have no way to stop the requests once the USD 200 are used up.

McGiverGim, 2019-02-23:
I think we can block it in the API configuration, but if they need to have a payment configured, is a bad idea. Maybe now the limit is 200$, but if they change it, maybe we can hit the limit before :-(

mikeller, 2019-02-23:
Even with the USD 200 limit giving us a seemingly large number of free API calls I do not think it is a risk we can afford to take - what if there was a bug that under certain circumstances had configurator go into a loop and try to load the map 100 times a second - leaving a single configurator instance in this state would chew through the free USD 200 in under 5 minutes...

McGiverGim, 2019-02-27:
Seeing the API for embed maps, that is what we I using (I think), it seems that the basic use, that is the use of the configurator, is free:
[image]

More info here: https://developers.google.com/maps/documentation/embed/guide#place_mode

What I don't know if it we can put in the console a limit of 0$ (to be protected if something goes wrong...)

mikeller, 2019-02-27:
@McGiverGim: Good point, I had not looked into the details of what requests are charged, but it looks like what we are doing might be available for zero charge.

This looks like it might be used to make sure it stays that way: https://cloud.google.com/billing/docs/how-to/budgets

It's not only that things can go wrong - since we'll have to put the API key into the application we ship, there is little we can do to stop people from extracting the key, and using it to access the API with apps other than the configurator...

McGiverGim, 2019-02-27:
Exactly! I don't know who can "officially" generate the key to test it. So you have access to it?

mikeller, 2019-02-27:
The way I understand it anybody can create an account with Google, then register his credit card, and then create an API key. I think before we jump into this we need to evaluate if this is what we want to do, as this puts a financial liability on us (even if we expect the cost to always be 0, this is still different from 'we are sure we won't have to pay, even if things go wrong).

laurentopia, 2019-02-27:
iNav uses OpenStreetMap. It works.

McGiverGim, 2019-02-27:
I have been playing today with OpenStreetMaps, I have a version more or less working, but then I noticed that I was using API version 2, and the latest is 5.3 :-P

I will try to upgrade...

closed 2019-03-03 by mikeller