betaflight/betaflight-configurator #1321: Pitch D Min value is  wrong

eumelOne, 2019-03-15:
The D Min value on the PIDs page is shown wrong. Value is "29" but shown as "0".  

[image]

[image]

McGiverGim, 2019-03-15:
I will look into it, thanks!

McGiverGim, 2019-03-16:
I've been looking at this and there is no bug. Your pitch D value is 28. D_Min must be lower than that, if not D_Min is deactivated, that is what is being shown in the tab.

eumelOne, 2019-03-16:
Seems pretty logical to me. My brain is out of order due to a cold... Sorry for stealing your time and thx for your help!

closed 2019-03-29 by McGiverGim