betaflight/betaflight-configurator #1328: Running multiple instances of Betaflight Configurator

andrejpodzimek, 2019-03-23:
Betaflight Configurator can't be run more than once. Multiple instances would be useful though, e.g. to quickly compare the configuration of mulitple FCs, to copy&paste between CLIs etc. This comes with the risk of opening the same `/dev/ttyX` from multiple Betaflights, but since nothing stops `betaflight-configurator` and `inav-configurator` from running side-by-side, two/multiple Betaflights may not be so bad.

mikeller, 2019-03-24:
@andrejpodzimek: This is not a limitation of the Betaflight configurator, this is a limitation of the NW.js framework that it is built upon. If you want this fixed, or want to help with fixing this, report an issue on https://github.com/nwjs/nw.js/issues.

andrejpodzimek, 2019-03-24:
I see. Closing this then. But still: Because I can run Betaflight, BLHeli and iNav at once, there’s probably an identifier of some sort that I could change to trick nwjs into believing that there are multiple different Betaflights… This could of course clobber some config files if not done carefully.

closed 2019-03-24 by andrejpodzimek

andrejpodzimek, 2019-03-24:
### I have multiple Betaflights running! :v:

Long story short, [on ArchLinux](https://aur.archlinux.org/packages/betaflight-configurator):

```
nw --user-data-dir=.config/betaflight-configurator-0 /usr/share/betaflight-configurator/
nw --user-data-dir=.config/betaflight-configurator-1 /usr/share/betaflight-configurator/
```

This^^^ has an obvious drawback of not sharing any stored configuration, but there's not exactly much of it anyway. So it won't remember what target I flashed last time &mdash; meh.

Why `--user-data-dir` didn't work form me before: Because the launcher script, [`/usr/bin/betaflight-configurator`](https://aur.archlinux.org/cgit/aur.git/tree/betaflight-configurator.sh?h=betaflight-configurator) doesn't care about command-line arguments. What it does:
```
nw /usr/share/betaflight-configurator/
```

What it should be doing:
```
nw /usr/share/betaflight-configurator/ "$@"
```

Distantly related issues are nwjs/nw.js/issues/6069, nwjs/nw.js/issues/6383, nwjs/nw.js/issues/6469.

In particular, nwjs/nw.js/issues/6383 is interesting: It suggests that there's a `single-instance` option. But the `single-instance` option [is actually deprecated](https://github.com/nwjs/nw.js/blob/1b88f7979b32c53c520ad886c34408e64cd01f8a/docs/For%20Users/Migration/From%200.12%20to%200.13.md#manifest-format), so that won't help.