betaflight/betaflight-configurator #1331: Small adjustment on Receiver tab

0crap, 2019-03-26:
Can it be made possible to add a value for "Derivative Cutoff Frequency" to be higher then 255 Hz?

The configurator currently does not allow values >255.

A value of 300 Hz is currently used when testing new features.
[image]

McGiverGim, 2019-03-26:
The Configurator uses the values that the MSP for the current version supports. Values higher of 255 maybe are added for Betaflight 4.1 but if I'm not wrong, in 4.0, that is the next release, 255 is the maximum.

0crap, 2019-03-26:
Fair enough!

closed 2019-05-02 by stale[bot]

0crap, 2019-05-15:
@McGiverGim Is it OK to reopen this issue?
4.0 is out the door and it would be great if this issue can be fixed on the next release.

McGiverGim, 2019-05-15:
At this moment the changes that incremented this value are not in master of Betaflight, if I'm not wrong. Are the changes in master?

mikeller, 2019-05-18:
Yes, this is still an `uint8_t` in the firmware:

https://github.com/betaflight/betaflight/blob/197796873d31bfebb25f305ecec962ca13a28c33/src/main/pg/rx.h#L58