betaflight/betaflight-configurator #1334: Feature Request: Mode toggle

skyfpv, 2019-03-30:
Introduce a feature which would allow a mode to be toggled on and off when the criteria are satisfied, as opposed to only being on when the criteria are satisfied and off when they are not. This would be useful for use with momentary switches to "activate" and "deactivate" a mode. It will also be really handy for radios with buttons like the Taranis X-Lite Pro.

McGiverGim, 2019-03-30:
I can't see the difference. You decide if you put the mode in the satisfied or not range to enable or disable it. Can you put an example?

McGiverGim, 2019-03-30:
Ah! You want tiggle the mode when pressing a button... Understood.

mikeller, 2019-03-30:
@skyfpv: I think what you are trying to do should rather be done on the TX side than on the flight controller: Just set up a 'sticky' logical switch with your button as the input and reset functions, and it will work just fine as an on / off flip flop switch:

[image]

Doing this on the flight controller will lead to unintended consequences, and potentially be dangerous: If there is a momentary loss of connection it will be impossible for the flight controller to determine the correct state of the mode after the connection is re-established, as it will have no way to know if the pilot has pushed the button in the mean time.

skyfpv, 2019-04-15:
@mikeller While I understand that this can be done radio side, I feel implementing it in software will be appreciated for users of more simplistic radios like the turnigy evolution for example. Also, there is no safety hazard which does not already exist, especially since the position of modes during RX loss can be set in the failsafe tab.

mikeller, 2019-04-15:
@skyfpv: 

> I feel implementing it in software will be appreciated for users of more simplistic radios like the turnigy evolution for example.

I am not quite sure what you mean by this. Looking at the Turnigy Evolution I can not see any buttons, only switches. Do you have examples of the transmitters you want this to work with?

closed 2019-05-22 by stale[bot]