betaflight/betaflight-configurator #1335: beta 10.5.0's "Modes" tab broken: Window not drawn when selected

RRAlex, 2019-03-31:
I used the following build of 10.5.0 to see how it would work with the pre-4.0 stock firmware onboard a new Mobula7 HD (Crazybee F4 FR):

https://ci.betaflight.tech/job/BetaFlight_Configurator/139/

The "Modes" tab's window wouldn't load passed the first line, or beyond the [ ARM ] left grey square and some outline on the right. 
But no lines or options and no sliders or data are showing anywhere and I tried multiple times. 
Works just fine in 10.4.0 though.

This is tested on Ubuntu 18.04.2 LTS.

Cheers!

McGiverGim, 2019-03-31:
Can you post an screen capture? Mine seems to work ok.

mikeller, 2019-03-31:
@RRAlex: Tested with CRAZYBEEF4FR, Betaflight 3.5.7, Betaflight configurator 10.5.0 (`master`). Mode tab is working just fine. Can you back up your flight controller config, and then try if resetting the firmware to defaults makes it go away?

Docteh, 2019-04-06:
@RRAlex Just to be clear, is this pre-4.0 like 3.5.7 or is the board shipping with a 4.0.0 version? can you give us the output of the version cli command?

PropBender, 2019-04-07:
I am seeing the same behavior (mostly blank and unusable Modes tab) on a couple of my boards.
Betaflight / OMNIBUS (OMNI) 4.0.0 Apr  7 2019 / 04:40:06 () MSP API: 1.41
Betaflight / FULLSPEEDF411 (FSD41) 4.0.0 Apr  6 2019 / 22:24:56 () MSP API: 1.41
The OMNIBUS hex was built using a modified target to fit the features I use.
The FULLSPEEDF411 hex file was built using a MATEKF411 target, modified to adjust gyro orientation.

Using the CLI interface and Configurator version 10.4.0 I was able to set all values, and have performed successful flights on both boards.  Full Chip erase was used on both firmware updates and multiple attempts were made. I also observed issues with the OSD tab and battery tab as well, which I will detail separately.

Using the same ChromeOS version 10.5.0 download of the configurator the following board appears to have no issues.
Betaflight / OMNIBUSF4SD (OBSD) 4.0.0 Mar 29 2019 / 23:32:43 (ef831e64b) MSP API: 1.41

mikeller, 2019-04-08:
@PropBender: See https://github.com/betaflight/betaflight-configurator/issues/1349#issuecomment-480777122.

afehners, 2019-04-11:
Same issue here with the released 10.5.0 of configurator.

Mobula 7
Betaflight / CRAZYBEEF3FS (CBFS) 4.0.0 Oct 13 2018 / 08:01:10 (3b479f92d) MSP API: 1.41

irg33k, 2019-04-11:
[image]

I'm using Betaflight 4.0 on a Crazybee F3 Pro from mobula 7. Also having the same issue here.

mikeller, 2019-04-11:
RRAlex: 

> I used the following build of 10.5.0 to see how it would work with the pre-4.0 stock firmware onboard a new Mobula7 HD (Crazybee F4 FR):

There is no way for Betaflight to make configurator work with unofficial builds of the firmware - either fix this by installing officially released firmware, or else ask your manufacturer to provide a configuration utility for the unofficial firmware that they supply.

closed 2019-04-11 by mikeller

RRAlex, 2019-04-13:
Sorry I couldn't come back earlier, but I got the same screenshot as above.
So I'll try 4.0 now that it's out and report back :)