betaflight/betaflight-configurator #1349: Blank OSD Configuration Tab

PropBender, 2019-04-07:
I have a couple of fc boards that are displaying blank OSD configuration tabs.
Using the most recent (10.5.0 Chromeos) configurator download from the Jenkins site, and custom builds based on a current download (4/6/2019) from the Betaflight GitHub.
Betaflight / OMNIBUS (OMNI) 4.0.0 Apr 7 2019 / 04:40:06 () MSP API: 1.41
Betaflight / FULLSPEEDF411 (FSD41) 4.0.0 Apr 6 2019 / 22:24:56 () MSP API: 1.41
The OMNIBUS hex was built using a modified target to fit the features I use.
The FULLSPEEDF411 hex file was built using a MATEKF411 target, modified to adjust gyro orientation.

Using the CLI interface and Configurator version 10.4.0 I was able to set all values, and have performed successful flights on both boards. Full Chip erase was used on both firmware updates and multiple attempts were made. 

Using the same ChromeOS Configurator and Betaflight download the following board appears to have no issues.
Betaflight / OMNIBUSF4SD (OBSD) 4.0.0 Mar 29 2019 / 23:32:43 (ef831e64b) MSP API: 1.41

mikeller, 2019-04-08:
Tested with Betaflight 4.0.0-RC6 on OMNIBUS and MATEKF411 and latest Betaflight configurator from `master` in ChromeOS:

All tabs working fine.

I suspect something went wrong when you built your custom targets - try again after installing an official version with 'Full Chip Erase' enabled.

McGiverGim, 2019-04-08:
Can you specify which features you remove/add from the compilation?

Docteh, 2019-04-08:
If the full speed board is just a MATEKF411 with the gyro chip in differently, maybe try RC6 build of MATEKF411 and then rotate the gyro and accelerometer on the setup tab?

PropBender, 2019-04-09:
My download was pulled from this level of GitHub.  "/betaflight/betaflight"
The resulting builds from that download demonstrated the compatibility issues that I mentioned.

After reading your comments and doing some more searching, I arrived at the following level and downloaded the archive. "/betaflight/betaflight/tree/4.0.0-RC6"
Building my targets from this download gives me a fully functional Configurator interface with both of the boards.

The 4.0.0-RC6 was what I wanted to build from, but I had difficulty finding it.

@Docteh your suggestion does appear to work, so there is really no need to build the custom file for the FULLSPEEDF411 board.

Thanks for the help and suggestions.

closed 2019-04-09 by PropBender

mikeller, 2019-04-09:
@PropBender: If you are building the custom firmware then It is recommended that, instead of downloading archives of the repository, you install git and use git (potentially with a frontend like https://tortoisegit.org/) to pull the version you want - this will quickly become more efficient than downloading a full archive for every new version.

PropBender, 2019-04-11:
@mikeller   Thanks for the tip, I'll check it out.  I'm still very much on the uphill slope of the learning curve.

dylanlikeavillan, 2019-04-23:
I'm having blank osd tab and blank modes tab with crazybee f4 v2 dx, receiver bound and showing in receiver tab, any resolution here

Docteh, 2019-04-23:
@dylanlikeavillan I've seen reports that the crazybee f4 board shipped with a development version of betaflight. your best bet would be to flash the release version of 4.0.0

dylanlikeavillan, 2019-04-24:
Thanks, can't seem to flash it either, I can connect and see live receiver
movement, just modes osd tabs are blank? Any ideas where to look to program
arm with cli commands?

Docteh, 2019-04-24:
I'll let you choose your adventure:

They explain the aux CLI command at the bottom of this page 
https://github.com/betaflight/betaflight/blob/master/docs/Modes.md

Or if you're on windows, and you want to get the DFU driver to work. 
https://github.com/betaflight/betaflight/wiki/Installing-Betaflight#dfu-flashing-under-windows---usb-dfu

dylanlikeavillan, 2019-04-24:
Thank you very much!