betaflight/betaflight-configurator #1351: set gyro_lowpass_type = PT1 shows as BIQUAD in configurator

0crap, 2019-04-08:
When I  do "set gyro_lowpass_type = PT1", save my changes and connect again the Configurator, it shows "Gyro Lowpass 1 Dynamic Filter Type" as BIQUAD. (Previously it showed PT1 as expected.)

If I set it, using the GUI to PT1, save and refresh, it's again back at BIQUAD. (Should be PT1.)

My guess is this bug might be related to PR #1339 ?

Running the Jenkins Chrome version with file dates from 4 april.

0crap, 2019-04-08:
[image]
[image]

mikeller, 2019-04-09:
@0crap: This should have been fixed with the changes in #1348, please try again with the latest configurator from `master`.

McGiverGim, 2019-04-09:
Yes, the static is disabled when the dynamic is enabled. No matter what value the static has.

0crap, 2019-04-09:
Thx for confirming, I wasn't sure if this was correct.