betaflight/betaflight-configurator #1356: Complete Nightmare DO NOT INSTALL 10.5 - Nothing Works! Back to 10.4 for me

BrigzyFPV, 2019-04-10:
I gave up when I could not get the Min Voltage, Min Warning etc to work, they are all at zero, you add in 2.8/4.5/2.9 and hit save and it zeros them all again. Modes page, completely blank apart from one half assed mode, complete waste of time. So many things are not working, I had to uninstall 10.5 and go back to 10.4 it's a complete waste of time, nobody can use this new one yet, **DO NOT INSTALL.**

mikeller, 2019-04-10:
@BrigzyFPV: What version of the Betaflight are you running on the flight controller you are trying to configure?

BrigzyFPV, 2019-04-10:
4.0 on a Mobula7 Whoop. It's terrible, so many things don't work, I don't know how anyone could use this configurator, the Modes Tab was blank for example.

mikeller, 2019-04-10:
@BrigzyFPV: 4.0 has not yet been released - try using officially released firmware, and things will work as expected.

BrigzyFPV, 2019-04-10:
Why does it have a pop up window on the 10.4 Configurator saying to install 10.5?

**Here's what it says:** 

**Notice**
You are using an outdated version of the Betaflight Configurator.
Version 10.5.0 is available online, please visit the release page to download and install the latest version with fixes and improvements.
Please close the configurator window before updating.

I understand what you are saying BUT, I am using BF 4.0 and it wasn't because I flashed my new quads with it, they came with BF 4.0. So what I am saying is if you are running BF 4.0 DO NOT install the new 10.5 Configurator because it doesn't work. New quads are arriving with BF 4.0 so if you update the configurator like it says to do you will not be able to use it on your new quads that came with BF 4.0 installed. I understand that the 10.5 configurator probably works with BF 3.5.1 for example but what are ppl meant to do if they update the configurator like it says and their new quads are running 4.0? They have to do what I did, install 10.5 and find out it doesn't work, then uninstall it, find the older version, download it, install it and you are back to where you were, a waste of time is what it is for new quad owners. I get what you are saying and it would make good sense if I had flashed and updated my quads to 4.0 but I didn't. The pop up window should be more specific, it should say, if you are running BF 4.0 (ie: you just purchased a new quad that came with it) do not install the new 10.5 configurator because it's not going to work.

mikeller, 2019-04-10:
@BrigzyFPV:

> I understand what you are saying BUT, I am using BF 4.0 and it wasn't because I flashed my new quads with it, they came with BF 4.0.

I can only repeat what I said before:

**Betaflight 4.0 has not been released yet.**

If a manufacturer sent you hardware with an inofficial build of an unreleased version of Betaflight on it, and it does not work with any official version of the Betaflight configurator, then you should ask the manufacturer to give you a configuration tool that works for the software version they are distributing, and not blame Betaflight for not supporting unofficial firmware.

> So what I am saying is if you are running BF 4.0 DO NOT install the new 10.5 Configurator because it doesn't work. 

Please stop spreading false information based on incorrect assumptions.

Closing this.

closed 2019-04-10 by mikeller

jtroutt19, 2019-04-10:
Working just fine for me maybe

BrigzyFPV, 2019-04-10:
Whatever, I am trying to help ppl so they don't waste their time, I am not blaming BF, I am simply saying that if you get a new quad with 4.0 it won't work so do not use the new config. Delete if you want, I am just trying to help ppl so they don't waste an hour of their life pissing around with it.

jtroutt19, 2019-04-10:
I got a eachine trashcan not long ago that had bf 4.0.0 on. I used configurator 10.5.0 to configure it. Didnt have any problems with it.

mikeller, 2019-04-10:
@BrigzyFPV: You are not helping anybody by spreading false and misleading information.