betaflight/betaflight-configurator #1358: BUG with 10.5.0 BF configurator

SyRexFPV, 2019-04-11:
Hi guys !
I'ved download today BF configurator 10.5.0 on windows 10 and I found a bug. 
When I go to "Modes" to check flight modes, windows don't open correctly and I can't go nowhere, I had to disconnect the FC.
My quad is a leader 120 stock (FullspeedRc) with a F3 FC.

SyRexFPV, 2019-04-11:
https://drive.google.com/file/d/1A32wqzlflAhiCgEQMpgojdA6bsNJGs3O/view?usp=sharing

Schmell, 2019-04-11:
Same here as above

SyRexFPV, 2019-04-11:
Really sorry ! I don't find it ! I m new on Github. Where is it ?

McGiverGim, 2019-04-11:
I understand that you use a Betaflight 4.0 NOT OFFICIAL firmware. The official firmware has been released today. If that is the case, ask the manufacturer of your quad or use the official version. We don't support firmware modified by others.

SyRexFPV, 2019-04-11:
Ok, no problem, I try to help. You right I use BF 4.0 not official firmware, it's BF 4.0 performance edition from Spatzengr https://github.com/spatzengr/betaflight/releases , special for F3 FC. I use it since 3.5 and usually, I had no problem with BF configurator. I'll use 10.4.0.

closed 2019-04-11 by SyRexFPV

McGiverGim, 2019-04-11:
You can talk with spatzenger and ask him to update the code to the latest version. This will work.

SyRexFPV, 2019-04-11:
Thanks. I'll try it.