betaflight/betaflight-configurator #1362: Modes tab doesn't display and hangs configurator.

uberjay2, 2019-04-12:
Just got a Mobula 7 and  thought I'd update to the latest BF Configurator.  Had 10.4.x and moved to 10.5.0.  Everything was fine until I clicked on the Modes tab.  All I get is the attached screen cap and it won't switch to any of the other tabs.  It does allow me to hit the disconnect button and after that I can reconnect and hit the other tabs (minus the Modes).  

I tried the new 10.5.1 release that just came out, but same issue.

For comparison, I fired up a local / old copy of BF 10.0.0 that I had sitting here and the modes tab works fine.

Let me know if you need more details / something executed on my PC.  I don't have a lot of time on my hands, but will do what I can to help.

Thanks!
[image]

McGiverGim, 2019-04-12:
Others users commented the same. 
Your mobula has not an official final Betaflight 4.0 version. Upgrade to the latest official firmware, or ask your manufacturer for a personalized Configurator.

uberjay2, 2019-04-12:
Thanks @McGiverGim ... I'll try that, but I find it surprising as I've never seen this before with any quads at different versions.  The rest of the UI works fine too which is surprising.  Either way, thanks.

uberjay2, 2019-04-12:
So I didn't try flashing the firmware... yet.  But what I did do is back level BF Configuration to 10.4.0.  It works fine with the same quad.  So I'm not sure why the firmware would be causing 10.5.x to puke on the modes tab??

Thanks,
J.

[image]

McGiverGim, 2019-04-12:
I'm not going to explain how the MSP protocol work with versions. I've explained your problem and the solution. You decide what to do.

uberjay2, 2019-04-12:
Okay, thanks for your time.  I wasn't trying to be a pest.... just didn't understand the details and looking at it at face value.  I'll close it off and upgrade the firmware when I can.  Thanks much.

closed 2019-04-12 by uberjay2

manjubssp, 2019-06-01:
Thanks. For the solution to my problem...

Arvenz200, 2020-01-28:
OK looks like i need to flash the firmware to the latest valid version. 

https://www.youtube.com/watch?v=slmUbB-dZsw 

Only when i try no response from bootloader. :( Help with that please?