betaflight/betaflight-configurator #1364: Add visualisation of rate_limit settings to PID Tuning tab graph

ianrmurphy, 2019-04-12:
Ref. https://github.com/betaflight/betaflight/issues/7982 if the rate_limit settings are added to MSP API v1.42 then it will be possible to visualise them on the graph on the PID Tuning tab.

Also see https://github.com/betaflight/betaflight/pull/6402

ianrmurphy, 2019-04-12:
I've already prototyped this feature, but since it needs co-ordination with any other MSP API v1.42 changes, I need to understand the collaboration process on the API change.

McGiverGim, 2019-04-12:
To see/modify them from the Configurator, they need to be added to the MSP protocol. The Configurator and the firmware only "speak" for this protocol.

Simplifying, you need to add them here first: https://github.com/betaflight/betaflight/blob/master/src/main/msp/msp.c

Usually, we have the feature at least one version only in CLI, and then we add it to MSP. I understand that your feature was included in Betaflight 4.0 (MSP API 1.41) so maybe you can add it in MSP in Betaflight 4.1 (MSP API 1.42).

ianrmurphy, 2019-04-12:
Yes I'd like to add to MSP API 1.42 - my concern is coordinating among any other changes to MSP API 1.42. If these are distributed among various pull requests, they need to sync up somehow. Is this done through some other channel? Or just in the natural sequence of merges?

McGiverGim, 2019-04-12:
No, you only need to make your part, the PR that adds the rate_limit values to some MSP command, and if accepted, it will be merged. You don't need to worry for the other part.

ianrmurphy, 2019-05-12:
Closed following merge.

closed 2019-05-12 by ianrmurphy