betaflight/betaflight-configurator #1365: Betaflight 10.5 configurator windows 10

chottofpv, 2019-04-12:
Not saving in pid tab. After I uncheck a tab to turn off the new yaw feature and hit save it automatically turns the feature back on. Also in the battery tab all the average , warning and minimal numbers are zeroed out . When I applied the right numbers for battery voltage and hit save they go back to zero. I am using a crazybee f4 board running betaflight 4.0 . Never seen the battery configuration tab numbers for stock settings all on zero and will not save once right values are entered. Also the new yaw feature can be checked off but once u save it automatically turns its self back on will not save . I went back to 10.4 configurator.

McGiverGim, 2019-04-12:
It seems that your quad has not an official final Betaflight 4.0 version. Upgrade to the latest official firmware, or ask your manufacturer for a personalized Configurator.

chottofpv, 2019-04-12:
Thank you for your response my friend but there is a bug in the
configurator. I am not the only one who is having these problems in the
battery tab and the modes tab .

McGiverGim, 2019-04-12:
Yes, and all the same origin. The final version of Betaflight 4.0 was published yesterday. Do you have this version? If not you have an unofficial version. The official version works with the Configurator.

chottofpv, 2019-04-12:
Ok thank u

chottofpv, 2019-04-16:
My friend this really late as we message almost a week ago but I just
wanted to be sure that the 10.5 configurator will only work with betaflight
4.0. Again thank you for your time

mikeller, 2019-04-16:
@chottofpv: In this case can you close this issue please?

closed 2019-04-16 by chottofpv